e-m-b-a/emba #1301: Support for Kali 2024.4

m-1-k-3, 2024-09-12:
From the 2024.3 release notes [here](https://www.kali.org/blog/kali-linux-2024-3-release/):

```
With Python 3.12, there will be a major change for users: it won’t be possible to install Python packages with pip anymore. [We wrote about that a year ago already](https://www.kali.org/blog/python-externally-managed/), we invite you to read that again if you are an avid user of pip.

But that will be for the next Kali release, 2024.4, due by the end of the year.With Python 3.12, there will be a major change for users: it won’t be possible to install Python packages with pip anymore. [We wrote about that a year ago already](https://www.kali.org/blog/python-externally-managed/), we invite you to read that again if you are an avid user of pip.

But that will be for the next Kali release, 2024.4, due by the end of the year.With Python 3.12, there will be a major change for users: it won’t be possible to install Python packages with pip anymore. [We wrote about that a year ago already](https://www.kali.org/blog/python-externally-managed/), we invite you to read that again if you are an avid user of pip.

But that will be for the next Kali release, 2024.4, due by the end of the year.
```

As EMBA is heavily using pip, we need to check the possibilities and start testing as soon as Python 3.12 is landed.

sagisar1, 2024-10-02:
I found a workaround, by using this command:
 rm /usr/lib/python3.12/EXTERNALLY-MANAGED
 
 it worked for me, as I don't use virtual environment and prefer using normal pip

m-1-k-3, 2024-10-23:
as we are currently on a good way to keep everything up with the current dev release, we can close this and will work with dedicated issues if needed

closed 2024-10-23 by m-1-k-3