e-m-b-a/emba #1310: Complex binary names aren't parsed correctly

levi-blodgett, 2024-09-18:
**Describe the bug**
When you have a binary that has more complex characters or parts than just `binary:version`, then EMBA doesn't fully know how to parse it.

i.e with a line like so in `s116_qemu_version_detection`:
`/usr/sbin/ntpdate;ntpdate;ntpdate 4.2.8p15;ntp:ntp:4.2.8:p15;unknown;1;`

The logic in EMBA will treat this not as an NTP binary of any kind, but like the binary `4.2.8` with a version of `p15`.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation of tag 1.4.2
2. Use the firmware available here: NTP should be available on most OpenWRT images
3. Start EMBA with the following parameters: Any, as long as it runs the package compilation & cve compilation modules
4. See bug

**Expected behavior**
It should be creating CVEs for the NTP binary, NOT a `4.2.8` binary that doesn't exist.

**Screenshots**
N/A

**Desktop (please complete the following information):**
- OS: Ubuntu 20.04, AWS EC2 instance
- EMBA version: v1.4.2
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - N

**Additional context**
N/A

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-09-19:
Thank you for reporting. I will try to reproduce it and come back to you later on

m-1-k-3, 2024-09-19:
Used this here https://downloads.openwrt.org/releases/23.05.4/targets/malta/be/openwrt-23.05.4-malta-be-default-rootfs.tar.gz but without _ntpdate_ in it. Which one can I use for testing?

PR #1323 (2024-10-15): Massive SBOM improvements
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Feature

* **What is the current behavior?** (You can also link to an open issue here)

Just little SBOMdorado

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

More SBOMdorado
closes #1310 
closes #1311 
closes #1333 
addresses also #443

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

Support the following sources for SBOM generation:
- [x] Binaries and librar [... cut]