e-m-b-a/emba #1311: How to reduce sbom detection time

chenjianquan7, 2024-09-19:
**Is your feature request related to a problem? Please describe.**
How to reduce sbom detection time
I just want to perform sbom detection and output sbom files, but its detection time is very long, especially for S09_firmware_mase-version_check, which takes a lot of time. Do I have any other ways to improve efficiency? Thank you

[image]

m-1-k-3, 2024-09-19:
If you are already using the sbom scan profile you could add the following to your profile `export QUICK_SCAN=1`. With this option only identifiers with known vulnerabilities are tested.
More CPU power is also a good solution for boosting up S09 :)

m-1-k-3, 2024-09-27:
please also check #1323

PR #1323 (2024-10-15): Massive SBOM improvements
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Feature

* **What is the current behavior?** (You can also link to an open issue here)

Just little SBOMdorado

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

More SBOMdorado
closes #1310 
closes #1311 
closes #1333 
addresses also #443

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

Support the following sources for SBOM generation:
- [x] Binaries and librar [... cut]
closed 2024-10-15 by m-1-k-3