e-m-b-a/emba #1313: docker compose dependency crash

qequ, 2024-09-21:
**Describe the bug**
Running docker-compose one of the dependencies being installed fails

```
1464.2       [ 10%] No update step for 'lief_libjson'
1464.2       [ 10%] No patch step for 'lief_libjson'
1464.2       [ 10%] No configure step for 'lief_libjson'
1464.2       [ 10%] No build step for 'lief_libjson'
1464.2       [ 10%] No install step for 'lief_libjson'
1464.2       [ 10%] Completed 'lief_libjson'
1464.2       [ 10%] Built target lief_libjson
1464.2       [ 10%] Generating internal/utfcpp/utf8.h
1464.2       [ 10%] Generating include/LIEF/third-party/internal/leaf.hpp
1464.2       [ 10%] Generating include/LIEF/third-party/internal/span.hpp
1464.2       [ 10%] Generating internal/nlohmann/json.hpp
1464.2       [ 10%] Building CXX object CMakeFiles/LIB_LIEF.dir/src/errors.cpp.o
1464.2       [ 10%] Building CXX object CMakeFiles/LIB_LIEF.dir/src/hash_stream.cpp.o
1464.2       [ 11%] Building CXX object CMakeFiles/LIB_LIEF.dir/src/logging.cpp.o
1464.2       [ 11%] Building CXX object CMakeFiles/LIB_LIEF.dir/src/exception.cpp.o
1464.2       In file included from /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/src/exception.cpp:19:
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:75:38: error: expected ‘)’ before ‘offset’
1464.2          75 |   explicit read_out_of_bound(uint64_t offset, uint64_t size);
1464.2             |                             ~        ^~~~~~~
1464.2             |                                      )
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:76:38: error: expected ‘)’ before ‘offset’
1464.2          76 |   explicit read_out_of_bound(uint64_t offset);
1464.2             |                             ~        ^~~~~~~
1464.2             |                                      )
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/src/exception.cpp:50:1: error: no declaration matches ‘LIEF::read_out_of_bound::read_out_of_bound(uint64_t, uint64_t)’
1464.2          50 | read_out_of_bound::read_out_of_bound(uint64_t offset, uint64_t size) : LIEF::exception("") {
1464.2             | ^~~~~~~~~~~~~~~~~
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:71:16: note: candidates are: ‘LIEF::read_out_of_bound::read_out_of_bound(LIEF::read_out_of_bound&&)’
1464.2          71 | class LIEF_API read_out_of_bound : public exception {
1464.2             |                ^~~~~~~~~~~~~~~~~
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:71:16: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound(const LIEF::read_out_of_bound&)’
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:71:16: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound()’
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:73:20: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound(const LIEF::exception&) [inherited from LIEF::exception]’
1464.2          73 |   using exception::exception;
1464.2             |                    ^~~~~~~~~
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:73:20: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound(std::string) [inherited from LIEF::exception]’
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:73:20: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound(const char*) [inherited from LIEF::exception]’
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:71:16: note: ‘class LIEF::read_out_of_bound’ defined here
1464.2          71 | class LIEF_API read_out_of_bound : public exception {
1464.2             |                ^~~~~~~~~~~~~~~~~
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/src/exception.cpp:58:1: error: no declaration matches ‘LIEF::read_out_of_bound::read_out_of_bound(uint64_t)’
1464.2          58 | read_out_of_bound::read_out_of_bound(uint64_t offset) : LIEF::exception("") {
1464.2             | ^~~~~~~~~~~~~~~~~
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:71:16: note: candidates are: ‘LIEF::read_out_of_bound::read_out_of_bound(LIEF::read_out_of_bound&&)’
1464.2          71 | class LIEF_API read_out_of_bound : public exception {
1464.2             |                ^~~~~~~~~~~~~~~~~
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:71:16: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound(const LIEF::read_out_of_bound&)’
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:71:16: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound()’
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:73:20: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound(const LIEF::exception&) [inherited from LIEF::exception]’
1464.2          73 |   using exception::exception;
1464.2             |                    ^~~~~~~~~
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:73:20: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound(std::string) [inherited from LIEF::exception]’
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:73:20: note:                 ‘LIEF::read_out_of_bound::read_out_of_bound(const char*) [inherited from LIEF::exception]’
1464.2       /tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/include/LIEF/exception.hpp:71:16: note: ‘class LIEF::read_out_of_bound’ defined here
1464.2          71 | class LIEF_API read_out_of_bound : public exception {
1464.2             |                ^~~~~~~~~~~~~~~~~
1464.2       make[3]: *** [CMakeFiles/LIB_LIEF.dir/build.make:138: CMakeFiles/LIB_LIEF.dir/src/exception.cpp.o] Error 1
1464.2       make[2]: *** [CMakeFiles/Makefile2:411: CMakeFiles/LIB_LIEF.dir/all] Error 2
1464.2       make[1]: *** [CMakeFiles/Makefile2:472: api/python/CMakeFiles/pyLIEF.dir/rule] Error 2
1464.2       make: *** [Makefile:306: pyLIEF] Error 2
1464.2       Traceback (most recent call last):
1464.2         File "<string>", line 2, in <module>
1464.2         File "<pip-setuptools-caller>", line 34, in <module>
1464.2         File "/tmp/pip-install-stold17x/lief_2909c0d7c52d43e3ab6911b51a450ede/setup.py", line 484, in <module>
1464.2           setup(
1464.2         File "/external/emba_venv/lib/python3.12/site-packages/setuptools/__init__.py", line 117, in setup
[... 56 lines cut ...]

```

**To Reproduce**
Steps to reproduce the behavior:
1. docker-compose build emba

**Expected behavior**
finishing the installation and running emba

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS:  Ubuntu 22.04 lts
- EMBA version:  master
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - N

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-09-22:
In the mean time please use the provided EMBA docker images.

m-1-k-3, 2024-09-26:
could you do a test of PR #1321

closed 2024-09-26 by BenediktMKuehne

m-1-k-3, 2024-09-26:
docker base image v1.4.2e is up and could be used for early testing

PR #1321 (2024-09-26): fix docker build
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

docker build failed - see #1313 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

should build again. - closes #1313 

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

* **Other information**: