e-m-b-a/emba #1333: Docker not installing

busby666, 2024-10-08:
Downloaded Kali Linux: kali-linux-2023.3-installer-amd64.iso
Set to 8 processors 16GB Memory
After initial install then did: apt full-update -y
Reboot

Did the GIT command to download the EMBA Installer.
./installer.sh -d

Once done I tried to do: ./emba -U

Get    the error: 
Some dependencies are missing - please check your installation
                                                                                                                                 
To install all needed dependencies, run 'sudo ./installer.sh'.                                                                   
Learn more about the installation on the EMBA wiki: https://github.com/e-m-b-a/emba/wiki/installation                            

Looks like docker is missing.

└─# apt install docker   
Package docker is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  wmdocker

Error: Package 'docker' has no installation candidate

m-1-k-3, 2024-10-09:
> Downloaded Kali Linux: kali-linux-2023.3-installer-amd64.iso

A one year old Kali as source could result in unexpected behavior. Could you do a test with the latest stable (2024.3)?

busby666, 2024-10-09:
The older version was previously what I used for EMBA as it worked.
Had tried the newer version earlier but did this again to try and make sure I did not mess up the steps.

Downloaded kali-linux-2024.3-installer-amd64.iso.
8 CPU
16GB Memory
100GB Drive

Install Kali with Basic Tools
Install Stacer
apt update
apt full-update -y
apt autoremove
reboot now

+-$ lsb_release -r
Release:        2024.3

cd Desktop
sudo -s
git clone https://github.com/e-m-b-a/emba.git
cd emba
sudo ./installer.sh -d
EMBA installation finished

./emba -U  
[*] Necessary utils on system:
    docker - not ok
    Missing docker - check your installation
    inotifywait - ok
    notify-send - ok

Some dependencies are missing - please check your installation
                                                                                                                  
To install all needed dependencies, run 'sudo ./installer.sh'.                                                    

+-# apt install docker         
Package docker is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  wmdocker

apt install wmdocker
./emba -U  

[*] Necessary utils on system:
    docker - not ok
    Missing docker - check your installation
    inotifywait - ok
    notify-send - ok

Some dependencies are missing - please check your installation

Tell me what I messed up or if there is anything I can do.

David

m-1-k-3, 2024-10-09:
is there no 'docker.io' package anymore in latest Kali?

m-1-k-3, 2024-10-09:
I need to perform some tests on it. On the other hand, the docs are showing the usual installation procedure: https://www.kali.org/docs/containers/installing-docker-on-kali/

busby666, 2024-10-09:
I was looking at the same URL for Kali in ref to docker.

However I did check this:
└─# apt install docker.io
docker.io is already the newest version (26.1.5+dfsg1-2+b1).

I verified the install and added the commands:
sudo systemctl enable docker --now
sudo usermod -aG docker $USER

Then removed the external directory and re-ran installer.sh -d

Again EMBA installed but then if I try and do the update I get the following errors.

[*] Elementary:
    user permission - ok
    Docker compose version - not ok
./helpers/helpers_emba_dependency_check.sh: line 210: docker: command not found
    Docker-compose EMBA image version - Updates available (docker base image v1.4.2e required)
    configuration directory - ok
    Python virtual environment - ok

[*] Necessary utils on system:
    docker - not ok
    Missing docker - check your installation
    inotifywait - ok
    notify-send - ok

Some dependencies are missing - please check your installation
                                                                                                                               
To install all needed dependencies, run 'sudo ./installer.sh'.                                                                 
Learn more about the installation on the EMBA wiki: https://github.com/e-m-b-a/emba/wiki/installation

m-1-k-3, 2024-10-09:
Ok, I can reproduce this issue

[image]

m-1-k-3, 2024-10-09:
could you try to install the package 'docker-cli' manually and re-run the installer afterwards?

busby666, 2024-10-09:
cd Desktop/emba
sudo -s
rm -R external
apt install docker-cli
apt install docker-doc
./installer.sh -d

NOW I do see the EMBA Docker files being downloaded.
run ./emba -U
Already up to date.
Using default tag: latest
latest: Pulling from embeddedanalyzer/emba
Digest: sha256:efe6701ccb710603f43aa0da098a5399814b84dd6ef337ed976a499ccee265c2
Status: Image is up to date for embeddedanalyzer/emba:latest
docker.io/embeddedanalyzer/emba:latest
[*] EMBA update - EPSS database update
Already up to date.
[*] EMBA update - CVE database update
Already up to date.
[*] EMBA update - docker image
Using default tag: latest
latest: Pulling from embeddedanalyzer/emba
Digest: sha256:efe6701ccb710603f43aa0da098a5399814b84dd6ef337ed976a499ccee265c2
Status: Image is up to date for embeddedanalyzer/emba:latest
docker.io/embeddedanalyzer/emba:latest

So looks like that worked.  Is there something else I can test before I start running a month long analysis?

================
One a personal note anything I can do to support EMBA I will try.
I can code; not sure I am at your level at all but happy to test anything you want.

m-1-k-3, 2024-10-10:
> So looks like that worked. Is there something else I can test before I start running a month long analysis?

looking good. EMBA is automatically performing a health check in the beginning. So, if this is passed everything should work. You are testing huge firmware images if they need a month long. Probably it makes sense to upgrade you EMBA hardware a bit ;) I am quite sure you are aware of it but with the default profile EMBA is using threading which will speed up the process massively ... you can also enable this via the cli parameter -t

> 
> One a personal note anything I can do to support EMBA I will try. I can code; not sure I am at your level at all but happy to test anything you want.

Wonderful :) We always need support on all edges of EMBA :) Just start with testing and finding bugs. You will find a mass of bugs ;) You can also walk through open issues and the contribution documentation [here](https://github.com/e-m-b-a/emba/blob/master/CONTRIBUTING.md) and [here](https://github.com/e-m-b-a/emba/blob/master/CONTRIBUTORS.md) for some inspiration. If you have some dedicated ideas just go ahead and open an issue for discussing and tracking your idea.

PR #1323 (2024-10-15): Massive SBOM improvements
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Feature

* **What is the current behavior?** (You can also link to an open issue here)

Just little SBOMdorado

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

More SBOMdorado
closes #1310 
closes #1311 
closes #1333 
addresses also #443

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

Support the following sources for SBOM generation:
- [x] Binaries and librar [... cut]
closed 2024-10-15 by m-1-k-3