e-m-b-a/emba #1334: Status not showing

busby666, 2024-10-11:
Using the latest version of both Kali and EMBA.

I had started an Analysis with 8 CPU and about 12GB of memory.  Sometime during the night system seemed to freeze so I forced a restart of the VM.

Once the system came up I was able to verify that the virtual disk was not full; still had 16GB left.

So I started the Analysis again but did not clear the LOG directory and tried to resume the analysis. Everything seemed to start fine. I did see task P99_prepare_analyzer again and that seemed to finish and right now the analysis is proceeding.

But noticed a minor issue at the bottom of the screen.  All of the status boxes are not populated and refreshing.  Not a major issue by any means.
[image]

m-1-k-3, 2024-10-11:
Thank you for reporting. Yea ... the restart feature is somehow not fully stable. I need to check on this soon

busby666, 2024-10-11:
I will try and do what I can to help I can certainly code but I am old C programmer.  However; I can code in most languages if I put my mind too it.  Just want to participate.  Maybe I can start with docs and notes.

busby666, 2024-10-11:
I seeing another odd issue but unable right now to gather information that would aid anyone in determining the problem.  VM just becomes unusable.

m-1-k-3, 2024-10-18:
I tried to reproduce this issue but for me it looks as it recovers the boxes quite good:

[image]

Could you please reproduce it on your side and give me step by step instructions?

torabi12, 2024-10-24:
I checked also and it works for me on Kali:

[image]

m-1-k-3, 2024-11-01:
Any updates on this? Otherwise I would close it.

busby666, 2024-11-01:
I think I closed it can you confirm?

closed 2024-11-01 by m-1-k-3