e-m-b-a/emba #1338: documentation about scan profiles

qequ, 2024-10-14:
**Is your feature request related to a problem? Please describe.**
hi, currently I'm using emba as default with no profiles but I've seen that there are already defined several profiles in `scan-profiles` but with no documentation about what options are set or skipped in each profile.

**Describe the solution you'd like**
wiki entries for the current scan profiles and what they do. Also a tutorial on how to make a custom scan profie  

**Describe alternatives you've considered**

**Priority issue**
Are you already a [Sponsor]? - N

**Additional context**
Add any other context or screenshots about the feature request here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-10-14:
If you check the default profile I think most of the options are explained or self-explaining:
https://github.com/e-m-b-a/emba/blob/master/scan-profiles/default-scan.emba

If something is unclear, please bring the option up and we can add a note.

qequ, 2024-10-30:
when running without the scan profile flag (-p) does emba use implicitly a scan profile? the default one?

m-1-k-3, 2024-10-30:
No, without a profile EMBA runs unconfigured and you need to configure it via parameters. At least you probably would use '-t' and '-W'. '-h' gives you more details on the possible parameters.

qequ, 2024-10-31:
if i set env variables without using a profile will that work ? like
```
$ export MODULE_BLACKLIST+=( "S10_binaries_basic_check" "S110_yara_check" "S12_binary_protection" "S13_weak_func_check" "S14_weak_func_radare_check" "S15_radare_decompile_checks" "S16_ghidra_decompile_checks" "S17_cwe_checker" "S99_grepit" )

$ sudo ./emba -f path/to/firmware  -l path/to/logs -t 
``` 

will the blacklist work ?

m-1-k-3, 2024-10-31:
No, currently we do not support environment variabls. The current approach is to use profiles or (where possible) command line parameters. I recommend to work with a scan-profile that you can adjust as you like

m-1-k-3, 2024-12-06:
Looks as nobody is working on it. I will close this for now. Please re-open it if needed

closed 2024-12-06 by m-1-k-3