e-m-b-a/emba #1340: emba stalls grepping

qequ, 2024-10-16:
**Describe the bug**
I'm currently running emba  in a VM with multithreading set (-t) and quick-scan profile and it stalls for a long time making greps  

hadn't checked the code but it seems that the helper  "S99_grepit" is the one that has a list of greps to make. it's weird that   S99_grepit is in MODULE_BLACKLIST for the quick_scan profile it runs anyway 

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
default

2. Use the firmware available here: https://www.supermicro.com/support/resources/getfile.php?SoftwareItemID=10007
3. Start EMBA with the following parameters: sudo ./emba.sh -f firmware/path -l /tmp/log/path -t -p ./scan-profies/quick-scan.emba

**Expected behavior**
it had finished. for a quick scan I thought it won't take more than a few hours to run.
Also s99_grepit runs even tough it's using the quick-scan profile

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Ubuntu 22.04.4 LTS
- EMBA version: current master
- Installation method: installation.sh by default

**Priority issue**
Are you already a [Sponsor]? - n

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2024-10-17:
Have seen another bug in our docker base image. I will fix the image and then we can check this issue

m-1-k-3, 2024-10-17:
Updates are up ... please do a test with the updated docker base image and the current master

qequ, 2024-10-18:
it worked faster with the new docker image and  updates thx :+1:

closed 2024-10-18 by qequ