e-m-b-a/emba #1342: Found during install

busby666, 2024-10-17:
Digest: sha256:bacb8a5f43d66cf655149349044501aa2b0011a996af50225ae9f72eb4374968
Status: Downloaded newer image for embeddedanalyzer/emba:1.4.2h                                                  
docker.io/embeddedanalyzer/emba:1.4.2h                                                                           
latest: Pulling from embeddedanalyzer/emba                                                                       
Digest: sha256:112abe8474442fba9c407070a3c07ca375776f2a2359a35b22a679f6df89375d                                  
Status: Image is up to date for embeddedanalyzer/emba:latest                                                     
docker.io/embeddedanalyzer/emba:latest                                                                           
./installer/I05_emba_docker_image_dl.sh: line 56: up: command not found                                          
------------------------------------------------------------------------------------------------                 
Error detected - status code 127                                                                                 
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"                                                      
Location:  ./installer/I05_emba_docker_image_dl.sh, line 56                                                      
Stack Trace:                                                                                                     
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 56 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 369 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

m-1-k-3, 2024-10-17:
have you performed another EMBA installation before on this host? I think there is an conflict between "docker-compose" and "docker compose". 

Could you check the following:
```
└─$ which docker-compose
└─$ dpkg -l | grep docker-compose

```
 and then check if your docker is already compose ready
```
└─$ sudo docker --help | grep compose
```

According to [here](https://docs.docker.com/engine/install/debian/) you should be able to clean your docker installation the following way:
```
 for pkg in docker.io docker-doc docker-compose podman-docker containerd runc; do sudo apt-get remove $pkg; done
```

busby666, 2024-10-17:
This was a NEW installation on a new VM.
=============================================================
+--(root?EMBA)-[/home/emba]
+-# which docker-compose
docker-compose not found
                                                                                                                                 
+--(root?EMBA)-[/home/emba]
+-# dpkg -l | grep docker-compose
ii  python3-compose                       1.29.2-6.3                               all          Python implementation of docker-compose file specification
                                                                                                                                 
+--(root?EMBA)-[/home/emba]
+-# docker --help | grep compose 
 
====================================================================
Ran the command:  for pkg in docker.io docker-doc docker-compose podman-docker containerd runc; do sudo apt-get remove $pkg; done

cd Desktop/emba
rm -r external

ran ./installer.sh -d
Docker daemon not running! Please check it manually and try again

if I now run ./emba -U I get docker nad EMBA updates reported and then the process terminated.
 To install all needed dependencies, run 'sudo ./installer.sh'.

Should be: ./installer.sh -d
RAN: apt install docker.io
rm -r external
re-ran ./installer.sh -d

I get the same error again
===================================================================
embeddedanalyzer/emba docker image
Description: EMBA docker images used for firmware analysis.
Download-Size : 5357 MB

docker.io and the EMBA docker image (if not already on the system) will be downloaded and installed!
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  python3-attr python3-certifi python3-compose python3-decorator python3-docker python3-dockerpty python3-docopt python3-dotenv
  python3-gssapi python3-invoke python3-json-pointer python3-jsonschema python3-jsonschema-specifications python3-nacl
  python3-packaging python3-paramiko python3-referencing python3-requests python3-rfc3987 python3-rpds-py python3-texttable
  python3-uritemplate python3-webcolors python3-websocket python3-yaml
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
8622
EMBA docker image will be downloaded.
CONTAINER VARIABLE SET TO embeddedanalyzer/emba:1.4.2h
1.4.2h: Pulling from embeddedanalyzer/emba
Digest: sha256:bacb8a5f43d66cf655149349044501aa2b0011a996af50225ae9f72eb4374968
Status: Image is up to date for embeddedanalyzer/emba:1.4.2h
docker.io/embeddedanalyzer/emba:1.4.2h
latest: Pulling from embeddedanalyzer/emba
Digest: sha256:112abe8474442fba9c407070a3c07ca375776f2a2359a35b22a679f6df89375d
Status: Image is up to date for embeddedanalyzer/emba:latest
docker.io/embeddedanalyzer/emba:latest
./installer/I05_emba_docker_image_dl.sh: line 56: up: command not found
------------------------------------------------------------------------------------------------
Error detected - status code 127
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/I05_emba_docker_image_dl.sh, line 56
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 56 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 369 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

------------------------------------------------------------------------------------------------

I can blow this away and do a new install if you want.

m-1-k-3, 2024-10-18:
> I can blow this away and do a new install if you want.

Nope ... we need to solve this issue. What is your exact base system? Kali Linux? Version 2024.3 or latest development? Ubuntu?

busby666, 2024-10-18:
Here is the ISO I used: kali-linux-2024.3-installer-amd64.iso

m-1-k-3, 2024-10-18:
After a plain installation of the Kali Linux system i did the following steps but I was not able to reproduce this behavior. Please check my steps and retry it yourself with the current dev fork:

```
└─$ git clone https://github.com/m-1-k-3/emba.git --branch multiple_updates

┌──(m1k3㉿kali)-[~]
└─$ cd emba 
                                                                                                                                                                                                                                            
┌──(m1k3㉿kali)-[~/emba]
└─$ sudo ./installer.sh -d
```

busby666, 2024-10-21:
That did seem to work; although I get new messages if I try the ./emba -U

Do you want me to try an Analysis?

m-1-k-3, 2024-10-21:
ok, perfect ... I will check the -U switch next

busby666, 2024-10-21:
[image]
[image]

Let me know if I can do anything.

m-1-k-3, 2024-10-21:
Yes because you are on my fork. Should be working on an installation from the original EMBA repo. Looks as we can close this issue.

closed 2024-10-21 by m-1-k-3

busby666, 2024-10-21:
OK I will do a new build and use the regular fork.  Thanks for the help.