e-m-b-a/emba #1348: Online check support repo returning empty hash values

secsonthebeach, 2024-10-21:
It appears an automated commit to  ‘onlinecheck’ support repo removed hash values from some files about 5 days ago.
On my end this is causing an “IndexError: list index out of range” error when attempting `run-server.sh` for Embark.

m-1-k-3, 2024-10-21:
Interesting ... I need to check on this.

secsonthebeach, 2024-10-21:
It appears to have been commit 0f823f211a92b326eb6aa703a4bbbec01163bfed 
Each of the files updated 5 days ago are now blank. 

https://github.com/EMBA-support-repos/onlinecheck/commit/0f823f211a92b326eb6aa703a4bbbec01163bfed

m-1-k-3, 2024-10-22:
should be fixed here https://github.com/EMBA-support-repos/onlinecheck/commit/c19e3fc57001b5434d0df18e2ba81802feed6962

m-1-k-3, 2024-10-22:
Thank you for reporting. Please check it and reopen if needed

closed 2024-10-22 by m-1-k-3

secsonthebeach, 2024-10-22:
That did the trick. Thank you!

m-1-k-3, 2024-10-22:
perfect

PR #1349 (2024-10-22): Fix update check
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

Update check is failing - see #1348 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closed #1348 

* **Other information**:

See also https://github.com/EMBA-support-repos/onlinecheck/commit/c19e3fc57001b5434d0df18e2ba81802feed6962