e-m-b-a/emba #1352: Final Report

busby666, 2024-10-22:
Final report in the INDEX.HTML
Last line of the report for some firmware indicates the following:

Remote exploits: 1 / Local exploits: 6 / DoS exploits: 1 / Github PoCs: 0 / Known exploited vulnerabilities: 7 / Verified Exploits: 0

However; unless the lines above this line is not clickable making it difficult to find the ONE remotely exploitable vulnerability.

very LOW priority but let me know if this should be a feature or something else.

m-1-k-3, 2024-10-23:
on the vulnerability aggregator (F20) you should be able to search for remote exploits via the indicator of capital R. Does this work for you?

busby666, 2024-10-23:
I see several files for F20* under the root of the log directory one ending in CSV.  When I open that file I do not see the R indicator.

I look F20_summary.csv did not have that either.  I did find it in the file: exploits-overview.txt so thank you for that I need to report to the vendor.  But was curious why there was no link to it.  if you want I can submit an Enhancement request?

closed 2024-10-23 by busby666

m-1-k-3, 2024-10-23:
I think exploits-overview is also included in the F20 web report.