e-m-b-a/emba #1355: Take infinitely long time and just doen't complete

controls37, 2024-10-26:
I am new to using EMBA and have been trying to experiment since last 1 week or so and my scans never finish.
Trying several firmware's some proprietary and some available for free.
Tried the proprietary initially, then I thought may be they are too big to analyse and I observed even after running the emba on them it did not complete for more than 2 days.

Command that I am using: (I want to use it in default state for now with no additional profile config)
- sudo ./emba -f ./<path-to-firmware-blog> -l ./<path-to-log-folder>

Initial configuration of execution environment:
- Ubuntu Cinnamon
- Host: Precision 7530
- RAM 64GB
- Intel R Xeon E-2176M
The size of the firmware was around 26GB of different file system merged into one file as that's how it is flashed on our target environment.
It tools more than 48 hours then I gave up and closed the tool.

Then i changed the firmware thinking that it could be because of the large size and used the one that's freely available here https://www.downloads.netgear.com/files/GDC/R6220/R6220-V1.1.0.114.zip but I still have the same problem.

Then I spawned an EC2 t2.xlarge with my own money.
And it has the exact same behavior for both the files.
The scan just keeps running forever.
So i do not think it's a problem of the hardware's capabilities.

I am attaching a log here, as I write this the stuff is still running since last more than an hour for the Netgear firmware file and hasn't ended. This is a really simple file and is only around 26MB in size.

[too-long.log](https://github.com/user-attachments/files/17529313/too-long.log)

controls37, 2024-10-26:
I let it run for 3+ hours on the EC2 large instance and then got tired and stopped it.
Here is the remaining log. It was so slow that it barely made 2-3 more items as can be seen from the log.
I don't know what wrong here.
Should I change the command or use a different profile ?

[too-long2.log](https://github.com/user-attachments/files/17529598/too-long2.log)

m-1-k-3, 2024-10-26:
Hi @controls37 

thank you for reaching us out. A 26GB firmware is very huge for EMBA. I think we had never tested something that big.
Nevertheless, if you want to use EMBA I suggest to optimize your command line a bit:
-t -> for threading. This is very important to allow EMBA to use your system resources effectively
-W -> for the final web reporter. Otherwise you will need to walk through the text logs.

Additionally, it proabably makes sense to not run all the modules. All these things are done in the profiles. I would suggest testing the quick-scan profile as starting point: https://github.com/e-m-b-a/emba/blob/master/scan-profiles/quick-scan.emba We have already blacklisted some long running modules https://github.com/e-m-b-a/emba/blob/447722647e1821e6eea708506c9e72ccc186b6a4/scan-profiles/quick-scan.emba#L29

Beside this, EMBA is not optimized for such big things. 26GB is probably to huge to finish in a useful time without further tweaking and more cores.

m-1-k-3, 2024-10-27:
I did a check with one of my virtual machines (8 cores and ~20Gig RAM) took ~10h while I was also working on this machine. 
I used the following EMBA startup command for this test:

```
└─$ sudo ./emba -f ~/Firmware_images/R6220-V1.1.0.114.zip -l ~/firmware-analysis/emba_logs_R6220 -y -p ./scan-profiles/default-scan.emba
```

controls37, 2024-10-30:
> I did a check with one of my virtual machines (8 cores and ~20Gig RAM) took ~10h while I was also working on this machine. I used the following EMBA startup command for this test:
>
> ```
> └─$ sudo [...]

10 Hours is a long time for this simple firmware. It's not feasible when it needs to be submitted with multiple firmware loads using distributed nodes in a staging environment that pushes the firmware to manual analysis mode.
I have started the scan with the changes you suggested along with the blacklisted modules for the "R6220-V1.1.0.114.zip" on my Xeon 64GB and Precision 12 core so let's see how long it takes now. It will populate the results and time when it completes.

Could you suggest why it takes so much time ? Is it because of the bash scripts ?

qequ, 2024-10-30:
hi, I recently had a similar issue with the duration of the analysis and not finishing and one of the main bottlenecks was an scan (S99_grepit) that makes greps searching for passwords and other stuff (https://github.com/e-m-b-a/emba/issues/1340). 
Checking with htop  seems that it runs a single grep or at most two or greps  at a time  and the list of greps is of several thousands. 
I thought this was fixed but yesterday started an analysis with other binary of a few megs and hasn't finished (more than 10 hours)

[image]

m-1-k-3, 2024-10-30:
> Could you suggest why it takes so much time ? Is it because of the bash scripts ?

Because it is doing some things for you like decompiling all binaries, doing static code analysis, building a SBOM on binary level and searching for passwords/hashes. There are a lot of things that need time on a broad base with lots of executables.

m-1-k-3, 2024-10-30:
btw there is also a quick scan profile available here https://github.com/e-m-b-a/emba/blob/master/scan-profiles/quick-scan.emba
Additionally, there was some blog post from Eclypsium where they discuss some optimizations. You can find it in our wiki here https://github.com/e-m-b-a/emba/wiki/Referring-sites-and-talks

controls37, 2024-11-04:
Thanks I've tried these methods you suggested but it doesn't finish in the time I need.
For now I am closing this issue as. May submit optimisations later in the future.

Thank you team. Keep up the good work.

closed 2024-11-04 by controls37

k4lizen, 2024-12-05:
Using [ripgrep](https://github.com/BurntSushi/ripgrep) when available instead of grep might speed things up considerably.

m-1-k-3, 2024-12-13:
> Thanks I've tried these methods you suggested but it doesn't finish in the time I need. For now I am closing this issue as. May submit optimisations later in the future.
> 
> Thank you team. Keep up the good work.

We did a bunch of updates which will results in faster tests.

m-1-k-3, 2025-01-28:
> Using [ripgrep](https://github.com/BurntSushi/ripgrep) when available instead of grep might speed things up considerably.

See https://github.com/e-m-b-a/emba/pull/1448