e-m-b-a/emba #1360: Error with Chat GPT Key

busby666, 2024-10-31:
Modified the config/gpt_config.env.template with the directions on page: https://github.com/e-m-b-a/emba/wiki/AI-supported-firmware-analysis

Then I copied the gpt_config.env.templage to gpt_config.env.

I attempted to test doing the command: ./emba -U

Then I get a TON of errors.
[-] ChatGPT error while testing the API-Key: KEYREDACTED
    OpenAI-API key  - not ok
[-] ChatGPT error while testing the API-Key: KEYREDACTED
    OpenAI-API key  - not ok
[-] ChatGPT error while testing the API-Key: KEYREDACTED
    OpenAI-API key  - not ok
[-] ChatGPT error while testing the API-Key: KEYREDACTED
    OpenAI-API key  - not ok
[-] ChatGPT error while testing the API-Key: KEYREDACTED
    OpenAI-API key  - not ok
[-] ChatGPT error while testing the API-Key: KEYREDACTED
    OpenAI-API key  - not ok
[-] ChatGPT error while testing the API-Key: KEYREDACTED
    OpenAI-API key  - not ok

Here is the .ENV File
OPENAI_API_KEY="KEYREDACTED"
GPT_QUESTION="For the following code I need you to tell me how an attacker could exploit it and point out all vulnerabilities."
MAXMIUM_GPT_PRIO=3

Do I have some error?  Do you have a better way for me to test and I will double check the API KEY.

busby666, 2024-10-31:
Additionally; if you do the ./emba -U when it gets to verify the  ChatGPT KEY it seems to be in an infinite loop.

busby666, 2024-10-31:
Did a TEST connection with Curl and get an error that the ONE call to test was too much.  I notice the model is GPT-40-mini.  Is there another I should test with?

BenediktMKuehne, 2024-10-31:
Model can be set under _config/gpt_template.json_ , that's the template used for openai-api calls.
Here are the model options: [https://platform.openai.com/docs/models](https://platform.openai.com/docs/models)

I'm asuming you tested with something like this: 
`curl -sS https://api.openai.com/v1/chat/completions -H "Content-Type: application/json" \
 -H "Authorization: Bearer ${OPENAI_API_KEY}" \
 -d @"${CONFIG_DIR}/gpt_template.json" --write-out "%{http_code}" -o /tmp/chatgpt-test.json`
              
What exactly was the error? Ratelimit? 
I haven't used this feature in a while, maybe they changed something

busby666, 2024-10-31:
The MODEL in the JSON file is gpt-3.5-turbo.  looks like according to the link you gave me the model might need to be: gpt-3.5-turbo-0125 but according to the documentation should work fine.

I typed the CURL command and get the same error:
==========================================================================
{
    "error": {
        "message": "You exceeded your current quota, please check your plan and billing details. For more information on this error, read the docs: https://platform.openai.com/docs/guides/error-codes/api-errors.",
        "type": "insufficient_quota",
        "param": null,
        "code": "insufficient_quota"
    }
}
429
======================================================

Tried a few other models and always get the same thing.

any suggestions; should we try grok; looks like the syntax would be similar?

BenediktMKuehne, 2024-11-02:
I have never used it.
Does it have advantages over open-ai?

m-1-k-3, 2024-12-04:
Changed the title as the current ChatGPT integration looks broken and is currently not working. Helpers are welcome!

m-1-k-3, 2025-01-17:
From my tests today it looks as you need to pay for API access now. Without payment I got the following error:

[image]

As soon as I payed 10$ for API access [here](https://platform.openai.com/settings/organization/billing/overview) it was working:

[image]

Also the EMBA dependency check switched to green:

[image]

I will update the docs that we need a payed API key for this feature.

m-1-k-3, 2025-01-17:
Wiki is updated here https://github.com/e-m-b-a/emba/wiki/AI-supported-firmware-analysis

closed 2025-01-17 by m-1-k-3

busby666, 2025-01-27:
Yea I was afraid of that.

Michael,

did notice some other things recently just reticent to put them on the GIT
without talking to you about it first.

I think I have time to try and reproduce the issue.  One of the things I
noticed is that I was doing the quick scan profile.  Sometimes my VM got
interrupted and I had to resume (not restart).  Then when I try to create a
tgz file out of the directory structure I got errors.   Even when as root I
tried to delete I got errors.  I had not seen this before.  if you want me
to try and re-create I will but it may take a week of running to see if I
can re-create the issue.  Then I am happy to give you everything I can.

David

m-1-k-3, 2025-01-28:
> I think I have time to try and reproduce the issue.  One of the things I
> noticed is that I was doing the quick scan profile.  Sometimes my VM got
> interrupted and I had to resume (not restart).  Then when I try to create a

Most of the time I have seen some weird behavior the system was running out of memory. You could check your system loogs for oom issues.

busby666, 2025-01-28:
Would it just be recommended to always RESTART and not try to resume?

m-1-k-3, 2025-01-28:
with restart you mean your OS? My system is restarted on every new install every year ;) In between I resume my VMs

busby666, 2025-01-29:
Michael,

Sorry by RESTART I mean an EMBA Scan. I would have the VM running on my
laptop and then sometime during the night I have a power outage or security
patch that reboots my system.

Once I get the VM up and running I can restart the EMBA Scan or RESUME;
seems like the resume was causing issues.

Even trying to TAR directories up was not working.

David

m-1-k-3, 2025-01-29:
Please check and if you can reproduce it please open a dedicated issue for this. Thank you