e-m-b-a/emba #1369: Analysis case fatal error

busby666, 2024-11-06:
Doing an analysis of an EXE.
at some point relatively quick a FATAL Crash occurs.  after reboot I get another error that a connection is refused and then another reboot.

I tried to rm -r external and do a re-install; this did not address the issue.

Very odd.  System comes up enter creds and get an error about not being able to get a connection.
Eventually it comes up and then reboots again.  If you tell me the logs to get I will try get the logs but systems seems to be stuck in some sort of reboot loop?

[image]

This is the only error I see.  Forced a complete Power OFF.  This seemed to stop the reboot cycle.  if you can tell me any logs that may help let me know.

m-1-k-3, 2024-11-07:
looks as something is broken on your machine. Was the system running out of memory or disk space? I would start with a check of the system logs.

busby666, 2024-11-07:
No was not running out of memory and diskspace.  I don't think I can give the binary.  If there is a specific log you want me to look at I can.  Not sure if it is worth the time though either.  Don't wish to waste time on something when I have so much other stuff to do like everyone else.

But I can repeat the problem so maybe I can add more switches to EMBA for logs or grab some of the logs from the EMBA Report.

David

m-1-k-3, 2024-11-07:
Without a testing firmware it is very hard for me ...

m-1-k-3, 2024-12-06:
Please re-open it with firmware to test. Close this for now