e-m-b-a/emba #1373: Inconsistent results with or without threading (-t) enabled and -mp61 -mp99 -ms09 modules

paianoa, 2024-11-10:
Emba 1.5.0

I noticed that running Emba against a firmware with options -mp61 -mp99 -ms09  I get different results if I enable Threading or not (-t option)

I would expected to have the same csv log files but this is not the case. I digged deep into the code and found out that PRE_THREAD_ENA variable of every P module is not honored when -m option is set with modules numbers too. This causes the P modules to not execute sequentially as in the case when no -t or -m options is provided (or yet -m options is provided with group module only:  -mp).

In my tests If I run with -t -mp61 -mp99 -ms09 I get different (and incorrect) results than if I run the same options without -t (correct results). This happens because, with -t option, RTOS variable, BINARIES, FILE_ARR are not detected completely before starting s09 module. S09 starts with the previous variable  set to default values.

Hope I explained it clearly and hope I'm not missing some design idea behind this behaviour

m-1-k-3, 2024-11-10:
The pre-checking modules are self organising and should not be configured with `-m` options. EMBA is enabling the module she needs automatically. Configuring these modules manually is currently not supported.

paianoa, 2024-11-10:
Thank you @m-1-k-3, didn't know about that. In that case It would helpful if EMBA prints an error message when P modules are provided with -m option. Feel free to close the issue.

m-1-k-3, 2024-11-10:
Good Idea. Coul you make a PR?

PR #1382 (2024-11-19): exit on pre-checking selection
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Feature

* **What is the current behavior?** (You can also link to an open issue here)

EMBA does not support manual pre-checker selection. If a user is trying to select pre-checking modules via -m parameter EMBA behaves weird. (see #1373 )

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

warn and exit - closes #1373

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to t [... cut]