e-m-b-a/emba #1384: Building from source - Emba

D3vil0p3r, 2024-11-21:
Emba can be only installed and used as a docker container or it can be built and used as a normal tool (non-containerized)?

m-1-k-3, 2024-11-21:
Hi @D3vil0p3r ,

Thank you for your request. It should be possible to install EMBA on a plain Kali without docker. Probably it will run into issues on BlackArch. You could try an installation via the `sudo ./installer.s -F` for our so called developer installation.

I'm quite sure this will fail as we never tested it. Nevertheless, if you are trying it and running into issues please let us know and we will try to support you.

m-1-k-3, 2024-11-21:
As EMBA is heavily optimized for debian based systems it probably will be a bit of a nightmare to port everything to Arch. Probably installing debs on Arch will be some kind of a helper. See https://unix.stackexchange.com/questions/83540/installing-a-deb-package-on-arch-is-it-possible

Would it be a solution to only install the stuff that is needed on the host and download the docker base image afterwards?

D3vil0p3r, 2024-11-21:
So all the dependencies are shown in bash files inside https://github.com/e-m-b-a/emba/tree/master/installer directory I guess.

m-1-k-3, 2024-11-21:
Yes, the installer scripts should include everything. (I added the download_file and curl command in [here](https://github.com/e-m-b-a/emba/issues/1384#issuecomment-2490649459)
Afterwards you should be able to use EMBA with the "emba" script from the repo. On a local installation you need to start it with `-D` parameter

D3vil0p3r, 2024-11-21:
Thanks for the information.

closed 2024-11-21 by D3vil0p3r

k4lizen, 2024-12-05:
Hey @D3vil0p3r have you created the package? It would be cool to have emba on the AUR

D3vil0p3r, 2024-12-05:
No. Currently it is more suitable to run as container.