e-m-b-a/emba #1391: Continuously failed during [+] I05_ emba_docker_image_dl

coderjingluo, 2024-11-28:
**Describe the bug**
Continuously failed during [+] I05_ emba_docker_image_dl

**To Reproduce**
Steps to reproduce the behavior:
1. sudo ./installer.sh -d or sudo -E ./installer.sh -d

**Expected behavior**
Continuously failed during [+] I05_ emba_docker_image_dl

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
- Kali Linux 2024.03
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

coderjingluo, 2024-11-28:
[image]

coderjingluo, 2024-12-03:
When enough docker image sources are added, sudo docker pull embeddedanalyzer/emba can be downloaded successfully, but what operations need to be performed after the download is successful to achieve the same effect as I05_emba_docker_image_dl? Do you need to execute sudo docker-compse build emba afterwards?

coderjingluo, 2024-12-03:
[image]
This is done by downloading sudo docker pull embeddedanalyzer/emba.

m-1-k-3, 2024-12-03:
Looks good. As EMBA is checking for a specific tag (see [docker-compose](https://github.com/e-m-b-a/emba/blob/8d232cd71487dbecd6c7f604821e54338e9b1a92/docker-compose.yml#L4)) you need to tag your base image now (see [here](https://github.com/e-m-b-a/emba/blob/8d232cd71487dbecd6c7f604821e54338e9b1a92/installer/I05_emba_docker_image_dl.sh#L54C11-L54C61)):
```
sudo docker tag embeddedanalyzer/emba:latest embeddedanalyzer/emba:1.5.0b
```
With this in place I think the installer should realize the image and should proceed

coderjingluo, 2024-12-03:
At this time, should I remove the I05 in the installer.sh file and run it? I tried it once, but there is no error in the back, but when I use emba, the report I get is empty

coderjingluo, 2024-12-03:
[image]
Run emba, then emba ends quickly and the report is empty
[image]
The virtual machine is configured as 4G and 2 core

m-1-k-3, 2024-12-03:
what says the docker logs monitoring?

coderjingluo, 2024-12-03:
docker logs emba_emba_run_fb6c14498db7 -f
[image]

[image]
docker logs emba_emba_quest_run_cfbabd6d8273 -f
[image]

m-1-k-3, 2024-12-03:
I have now found the issue with the wrong docker base image. The latest tag is not pointing to the correct image. I will fix this,

coderjingluo, 2024-12-03:
When using docker to pull, you must specify the version, otherwise it will pull to other versions of the image.

m-1-k-3, 2024-12-03:
latest and 1.5.0b should now reference the same docker base image

psparc82, 2024-12-05:
same issue. Deleting the old docker images and running

```
sudo docker pull embeddedanalyzer/emba:1.5.0b
sudo docker tag embeddedanalyzer/emba:1.5.0b embeddedanalyzer/emba:latest
```

and then running `sudo ./installer.sh -d` gives the same error. Is this the correct order of commands I need to run and will this break future updates as I've tagged v1.5.0b as the latest but it won't be the latest in future releases?

m-1-k-3, 2024-12-05:
I do not know what is the issue with your docker environment. As you are able to download the docker image manually I would go ahead and comment the following line

https://github.com/e-m-b-a/emba/blob/156a6709b55bf0e8511066012b0c511b7dbf26d7/installer.sh#L369

With this the installer should proceed.

m-1-k-3, 2024-12-15:
Installation finished?

m-1-k-3, 2024-12-30:
please reopen if needed