e-m-b-a/emba #1400: Issue faced while upgrading and installing the emba

ChethanPuranic, 2024-12-06:
**Describe the bug**
Issue faced while upgrading and installing the emba

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. Use the firmware available here: .upk file
3. Start EMBA with the following parameters: sudo ./installer.sh -d
4. additional steps
5. See error

(emba_venv) root@VM-test:~/embark/emba/new_emba# sudo ./installer.sh -d

EMBA - Embedded Linux Analyzer Installer
=================================================================

==> Imported 24 installer module files

Install all dependecies for EMBA in default/docker mode

Created external directory: ./external

Update package lists.
Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:2 https://download.docker.com/linux/ubuntu focal InRelease                                                                                   
Hit:3 http://us.archive.ubuntu.com/ubuntu jammy InRelease                                                                                        
Hit:4 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease                                                          
Hit:5 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease                 
Ign:6 http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 InRelease           
Hit:7 http://us.archive.ubuntu.com/ubuntu jammy-security InRelease                  
Get:8 http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release [3462 B]
Get:9 http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release.gpg [801 B]
Hit:10 http://cz.archive.ubuntu.com/ubuntu jammy InRelease
Ign:9 http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release.gpg
Reading package lists... Done
W: https://download.docker.com/linux/ubuntu/dists/focal/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
W: http://repo.mongodb.org/apt/ubuntu/dists/xenial/mongodb-org/3.2/Release.gpg: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
W: GPG error: http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release: The following signatures were invalid: EXPKEYSIG D68FA50FEA312927 MongoDB 3.2 Release Signing Key <packaging@mongodb.com>
E: The repository 'http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
------------------------------------------------------------------------------------------------
Error detected - status code 100
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer.sh, line 280
Stack Trace:
    [1] main(): ./installer.sh, line 280 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

------------------------------------------------------------------------------------------------

I have followed the below steps to resolve the issue but it didn't help
https://stackoverflow.com/questions/34733340/mongodb-gpg-invalid-signatures 

Please let me know how to resolve the issue.

m-1-k-3, 2024-12-06:
As EMBA is not using and installing Mongodb anymore this issue is very interesting. Probably this is from a very old installation?!? The following steps will help in cleaning up the system:
```
/etc/init.d/mongod stop
systemctl stop mongod
apt-get purge mongodb-org -y
rm /etc/apt/sources.list.d/mongodb-org-4.4.list
```

ChethanPuranic, 2024-12-08:
(emba_venv) root@Embark31:~# /etc/init.d/mongod stop
-bash: /etc/init.d/mongod: No such file or directory
(emba_venv) root@Embark31:~# systemctl stop mongod
(emba_venv) root@Embark31:~# apt-get purge mongodb-org -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package mongodb-org
(emba_venv) root@Embark31:~# rm /etc/apt/sources.list.d/mongodb-org-4.4.list
rm: cannot remove '/etc/apt/sources.list.d/mongodb-org-4.4.list': No such file or directory
(emba_venv) root@Embark31:~# 

This is the result I am getting. 
-----------------------------------------------
I tried to install again but same result.
(emba_venv) root@Embark31:~/embark/emba/new_emba# sudo ./installer.sh -d

EMBA - Embedded Linux Analyzer Installer
=================================================================

==> Imported 24 installer module files

Install all dependecies for EMBA in default/docker mode

Created external directory: ./external

Update package lists.
Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:2 https://download.docker.com/linux/ubuntu focal InRelease                                                                                                   
Hit:3 http://us.archive.ubuntu.com/ubuntu jammy InRelease                                                                                                                  
Ign:4 http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 InRelease                                                                                                  
Hit:5 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:6 http://cz.archive.ubuntu.com/ubuntu jammy InRelease
Hit:7 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease
Get:8 http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release [3462 B]
Hit:9 http://us.archive.ubuntu.com/ubuntu jammy-security InRelease
Get:10 http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release.gpg [801 B]
Ign:10 http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release.gpg
Reading package lists... Done
W: https://download.docker.com/linux/ubuntu/dists/focal/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
W: http://repo.mongodb.org/apt/ubuntu/dists/xenial/mongodb-org/3.2/Release.gpg: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
W: GPG error: http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release: The following signatures were invalid: EXPKEYSIG D68FA50FEA312927 MongoDB 3.2 Release Signing Key <packaging@mongodb.com>
E: The repository 'http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
------------------------------------------------------------------------------------------------
Error detected - status code 100
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer.sh, line 280
Stack Trace:
    [1] main(): ./installer.sh, line 280 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

------------------------------------------------------------------------------------------------

m-1-k-3, 2024-12-08:
You have some mongodb configuration in apt:

```
E: The repository 'http://repo.mongodb.org/apt/ubuntu xenial/mongodb-org/3.2 Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

This needs to be removed.

ChethanPuranic, 2024-12-10:
Below solution helped for me.
https://askubuntu.com/questions/842592/apt-get-fails-on-16-04-or-18-04-installing-mongodb 
Now installation completed successfully and able to run the scan.
Thank you.

m-1-k-3, 2024-12-15:
Thank you for the details. I will close this issue now

closed 2024-12-15 by m-1-k-3