e-m-b-a/emba #1414: unable to install EMBA Version 1.5.0c

busby666, 2024-12-17:
**Describe the bug**
Doing a new install.  Used Kali Image kali-linux-2024.3-installer-amd64.iso
8 Processors
8GB Memory
130GB Drive

After the initial install of Kali.

I removed the following: apt remove nmap burpsuite sqlmap john hydra netexec aircrack-ng metasploit-framework wireshark tcpdump responder atril*

Then changed to cd Desktop
git clone https://github.com/e-m-b-a/emba.git
cd emba
sudo -s
sudo ./installer.sh -d

I get the following error after downloading and extracting the containers.
EMBA docker image will be downloaded.
CONTAINER VARIABLE SET TO embeddedanalyzer/emba:1.5.0c
1.5.0c: Pulling from embeddedanalyzer/emba
Digest: sha256:80b537e2133c1e81b1e4b4c0225f19753c7ade35372db93980107d8fd7398d54
Status: Image is up to date for embeddedanalyzer/emba:1.5.0c
docker.io/embeddedanalyzer/emba:1.5.0c
./installer/I05_emba_docker_image_dl.sh: line 57: up: command not found
------------------------------------------------------------------------------------------------
Error detected - status code 127
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/I05_emba_docker_image_dl.sh, line 57
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 57 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 369 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

==================
I know Kali Released a new version so then I did an apt update & apt full-upgrade -y & reboot now

Logged back in.
rm -r external

Then redid the installer; but I get the same error.  

Seems to be reporting on line 57 in the script but not sure the line number is correct I see nothing wrong with command:
   "${DOCKER_COMPOSE[@]}" up --no-start

**Expected behavior**
Expected the install to work so must be a new version.

**Screenshots**
attached
[image]

m-1-k-3, 2024-12-17:
looks as an error in the detection of your docker environment. Please check if you have `docker-compose` or `docker compose` avaiable on your system.

busby666, 2024-12-17:
└─# apt install docker-compose
Package docker-compose is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Error: Package 'docker-compose' has no installation candidate

m-1-k-3, 2024-12-17:
Just hit 'docker compose' (with space) and check If there is an error.

busby666, 2024-12-17:
docker compose
docker: 'compose' is not a docker command.

I did a docker help and do not see a reference to this either.

if you want me to do something let me know I can do anything to this env.

m-1-k-3, 2024-12-17:
ok, looks as you have an old docker installation on your system. You can find the installation instructions for the newer docker version in the installer.sh [here](https://github.com/e-m-b-a/emba/blob/767631142315f462bfc0641504317c9f575faa1e/installer.sh#L294). I would recommend to uninstall your installed docker packages before.

```
  export OS="debian"
  apt-get install -y ca-certificates curl gnupg
  install -m 0755 -d /etc/apt/keyrings
  curl -fsSL https://download.docker.com/linux/"${OS}"/gpg -o /etc/apt/keyrings/docker.asc
  chmod a+r /etc/apt/keyrings/docker.asc
  echo "deb [arch=$(dpkg --print-architecture) signed-by=/etc/apt/keyrings/docker.asc] https://download.docker.com/linux/${OS} \
    bookworm stable" | tee /etc/apt/sources.list.d/docker.list > /dev/null
  fi
  apt-get update -y
  apt-get install -y docker-ce docker-ce-cli containerd.io docker-buildx-plugin docker-compose-plugin
```

busby666, 2024-12-17:
did an apt remove docker* -y
Then did the steps above with the exception of the FI as I did not see the start of an if block.

rm -r external

 ./installer.sh -d

Looks like that worked.  Trying an Analysis now.

And thanks for all of the support.

m-1-k-3, 2024-12-17:
ok, great. Yes, the fi was a c&p failure.
I will take another look at the installation routine to handle such situations better.

busby666, 2024-12-17:
Like said anything I can do to help please let me know.  I can script/code to a degree so would like to help if I can.

m-1-k-3, 2024-12-17:
If you can reproduce the failing setup it would be great if we can debug and fix it together

busby666, 2024-12-18:
Soon as I finish the current analysis; I will record step by step and send
via email here is that works?

I have a LIST of steps I use and update every time I rebuild, many are just
mine but some seem to make sure things work.  I would be pleased to be able
to assist.

David

busby666, 2024-12-19:
Do you want a bug report on this?
[image]

busby666, 2025-01-06:
> Absolutely

Michael,

Have not forgotten about you.  Analysis is still running.

m-1-k-3, 2025-02-13:
I will close this for now. Re-open if needed

closed 2025-02-13 by m-1-k-3