e-m-b-a/emba #1422: Ubuntu 22.04.4 LTS  FIX helpers.sh

tttttint, 2024-12-30:
ERROR
[image]
jq  
Description: lightweight and flexible command-line JSON processor  
Download-Size: 52.5 kB  
------------------------------------------------------------------------------------------------  
Error detected - status code 1  
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"  
Location:  ./installer/helpers.sh, line 68  
Stack Trace:  
    [1] print_tool_info(): ./installer/helpers.sh, line 68 -> print_tool_info jq 1  
    [2] I01_default_apps_host(): ./installer/I01_default_apps_host.sh, line 24 -> I01_default_apps_host  
    [3] main(): ./installer.sh, line 345 -> main -d  

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues  

以下是一篇完整的 GitHub Issue 模板，描述了问题、解决方案以及一些相关建议：

Title:

Error detected in helpers.sh when running jq - Missing jq installation and Docker setup clarification

Description:

I encountered the following error when running the script:

jq  
Description: lightweight and flexible command-line JSON processor  
Download-Size: 52.5 kB  
------------------------------------------------------------------------------------------------  
Error detected - status code 1  
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"  
Location:  ./installer/helpers.sh, line 68  
Stack Trace:  
    [1] print_tool_info(): ./installer/helpers.sh, line 68 -> print_tool_info jq 1  
    [2] I01_default_apps_host(): ./installer/I01_default_apps_host.sh, line 24 -> I01_default_apps_host  
    [3] main(): ./installer.sh, line 345 -> main -d  

---------------
Steps to Reproduce:
	1.	Run the script on Ubuntu 22.04.4 LTS.
	2.	Observe the error message indicating a missing jq dependency.

Cause:

The error occurs because the system does not have jq installed, which is required for processing JSON data in the script.

Additionally, some users might face issues with Docker and Docker Compose setup, which are prerequisites for the tool.

Solution:

1. Install jq:

Ensure that jq is installed on your system. Use the following command:

sudo apt update && sudo apt install jq -y

2. Install Docker and Docker Compose:

Follow the steps below to set up Docker and Docker Compose on Ubuntu properly:

Step 1: Install Docker

# Update the package index
sudo apt update

# Install Docker's package dependencies
sudo apt install apt-transport-https ca-certificates curl software-properties-common -y

# Add Docker's official GPG key
curl -fsSL https://download.docker.com/linux/ubuntu/gpg | sudo gpg --dearmor -o /usr/share/keyrings/docker-archive-keyring.gpg

# Add Docker's official repository to the APT sources
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/docker-archive-keyring.gpg] https://download.docker.com/linux/ubuntu $(lsb_release -cs) stable" | sudo tee /etc/apt/sources.list.d/docker.list > /dev/null

# Install Docker
sudo apt update
sudo apt install docker-ce docker-ce-cli containerd.io -y

# Verify Docker installation
sudo docker --version

Step 2: Install Docker Compose

# Download the latest version of Docker Compose
sudo curl -L "https://github.com/docker/compose/releases/download/$(curl -s https://api.github.com/repos/docker/compose/releases/latest | jq -r .tag_name)/docker-compose-$(uname -s)-$(uname -m)" -o /usr/local/bin/docker-compose

# Apply executable permissions to the binary
sudo chmod +x /usr/local/bin/docker-compose

# Verify Docker Compose installation
docker-compose --version

Step 3: Enable and Start Docker

# Enable Docker to start on boot
sudo systemctl enable docker

# Start Docker
sudo systemctl start docker

3. Update the Script:

Ensure your script includes checks for the presence of jq, and gracefully handles the error if it’s missing.

Add this to the script before invoking jq:

# Check if jq is installed
if ! command -v jq &> /dev/null; then
    echo "Error: jq is not installed. Please install jq using 'sudo apt install jq' and try again."
    exit 1
fi

Expected Behavior:

With jq installed and Docker/Docker Compose correctly set up, the script should run without errors and process JSON data as expected.

Environment:
	•	OS: Ubuntu 22.04.4 LTS
	•	Docker Version: Ensure the latest stable release.
	•	Docker Compose Version: Ensure the latest stable release.

Additional Notes:

If you encounter any further issues, feel free to reply to this issue, and I’ll assist you further!

--------

FIX helpers.sh

print_tool_info(){
  echo -e "\\n""${ORANGE}""${BOLD}""${1:-}""${NC}"
  TOOL_INFO="$(apt-cache policy "${1:-}" 2> /dev/null)"
  if echo "${TOOL_INFO}" | grep -q "已安装："; then
    INSTALLED_VERSION=$(echo "${TOOL_INFO}" | grep "已安装：" | awk '{print $2}')
    CANDIDATE_VERSION=$(echo "${TOOL_INFO}" | grep "候选：" | awk '{print $2}')
    if [[ "${INSTALLED_VERSION}" == "${CANDIDATE_VERSION}" ]]; then
      echo -e "${GREEN}""${1:-}"" is already installed and up-to-date.""${NC}"
    else
      echo -e "${ORANGE}""${1:-}"" is installed but will be updated.""${NC}"
      INSTALL_APP_LIST+=("${1:-}")
    fi
  else
    echo -e "${ORANGE}""${1:-}"" will be newly installed.""${NC}"
    INSTALL_APP_LIST+=("${1:-}")
  fi
}

----------

m-1-k-3, 2025-01-08:
Sorry for the delay.

Your first issue was around the docker installation. Your system has already a docker-compose installation and the installer complains on it. You can uninstall your current docker enviroment and then the installer should be able to handle the fresh installation. The docker installation routine can be found [here](https://github.com/e-m-b-a/emba/blob/dbd58608b25f8f91805636273209f3c937d9169b/installer.sh#L289)

The second issue is weird as jq should be installed [here](https://github.com/e-m-b-a/emba/blob/dbd58608b25f8f91805636273209f3c937d9169b/installer/I01_default_apps.sh#L24)

m-1-k-3, 2025-02-13:
please reopen it if needed

closed 2025-02-13 by m-1-k-3