e-m-b-a/emba #1458: Installer Fails When Removing Docker-Compose

max-wyze, 2025-02-11:
# Bug Report: Installer Fails When Removing Docker-Compose

## Describe the Bug

When installing the new version of EMBA, the installer detects an existing docker-compose installation and recommends that it be removed. However, if docker-compose is removed before running the installer, the installation fails with errors.

**Error when docker-compose is removed:**

```plaintext
Activating Python Environment - ./external/emba_venv

WARNING: Old docker-compose installation found
It is recommend to remove the current installation and restart the EMBA installation afterwards!

After removing docker-compose and running the installer, the installation fails. To work around the issue, I reinstalled docker-compose and then ran the installer successfully.

Installer output after reinstalling docker-compose:

```
# cd /opt
# git clone https://github.com/e-m-b-a/emba.git
# cd emba
# sudo ./installer.sh -d
linux-modules-extra-6.8.0-1018-aws
Description: Linux kernel extra modules for version 6.8.0 on 64 bit x86 SMP
Download-Size: 70.8 MB
linux-modules-extra-6.8.0-1018-aws won't be updated.
[*] Testing for unattended update settings

Automatic updates are enabled - this could result in unexpected behavior during installation!
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.

[+] I01_default_apps
=================================================================

[+] I13_disasm
=================================================================

[+] I05_emba_docker_image_dl
=================================================================

embeddedanalyzer/emba docker image
Description: EMBA docker images used for firmware analysis.
Download-Size : 5610 MB

docker.io and the EMBA docker image (if not already on the system) will be downloaded and installed!
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
730
EMBA docker image will be downloaded.
CONTAINER VARIABLE SET TO embeddedanalyzer/emba:1.5.1b
1.5.1b: Pulling from embeddedanalyzer/emba
Digest: sha256:30e52933857ec4d13e98a066e23bbe91a46ce5450368429cdd846ea68a114a9e
Status: Image is up to date for embeddedanalyzer/emba:1.5.1b
**docker.io/embeddedanalyzer/emba:1.5.1b
./installer/I05_emba_docker_image_dl.sh: line 57: up: command not found
------------------------------------------------------------------------------------------------
Error detected - status code 127
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/I05_emba_docker_image_dl.sh, line 57
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 57 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 369 -> main -d
```

**To Reproduce**
	1.	Run the EMBA installer (either in dev-mode or default mode).
	2.	Remove the existing docker-compose installation when prompted.
	3.	Execute the installer again:

apt remove docker-compose
sudo ./installer.sh -d

**Observe the failure.**

Note: Reinstalling docker-compose before running the installer allows the installation to complete successfully.

**Expected Behavior**

The installer should either:
	•	Handle the removal of docker-compose gracefully without failing, or
	•	Provide a clear, actionable error message if docker-compose must remain installed during the installation process.

**Desktop**
	•	OS:#19~22.04.1-Ubuntu SMP Wed Oct  9 16:48:22 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
	•	EMBA version: current master branch
	•	Installation method: Default installation using the up-to-date docker image

**Priority Issue**

Are you already a [Sponsor](https://github.com/sponsors/e-m-b-a)? (Y/N)
No

m-1-k-3, 2025-02-11:
looks as there is no `docker` installation performed.

m-1-k-3, 2025-03-08:
Can you give me a quick overview of the steps to reproduce this issue?

m-1-k-3, 2025-03-08:
Ok, got it. Looks as it is not only docker-compose to remove. After removing the complete docker installation the installer was able to move on:

```
m1k3@m1k3:~/emba$ sudo apt-get remove docker docker-compose docker.io python3-docker python3-dockerpty 
```
I will update the message from the installer in the next PR.

Thank you for reporting this issue.

PR #1500 (2025-03-09): No findings no log/Installer warning
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

see #1458

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closes #1458

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

* **Other information**:
closed 2025-03-09 by m-1-k-3