e-m-b-a/emba #1464: Race condition in P99 / helpers_emba_prepare.sh

gluesmith2021, 2025-02-14:
**Describe the bug**

A race condition exists in `helpers_emba_prepare.sh`:`binary_architecture_threader()`, indirectly called from P99 module, that can make MD5 sums and their associated file non-unique in `cvs_logs/p99_prepare_analyzer.csv`. 

```sh
  # Race occurs when two process checks for the same MD5 sum at once here, and it is not yet in p99_md5sum_done.tmp.
  # Both process will continue.
  if grep -q "${lMD5SUM}" "${TMP_DIR}/p99_md5sum_done.tmp" 2>/dev/null; then
    return
  fi

  # Both process add the same MD5 sum and keep processing the current file until it is added to p99_prepare_analyzer.csv
  echo "${lMD5SUM}" >> "${TMP_DIR}/p99_md5sum_done.tmp"
```

This can occurs when a firmware contains duplicated files with different names, and likelihood increases when they are in the same folder with similar names (i.e. when they are likely to be process one right after the other).

This non-uniqueness may have unintended consequences wherever uniqueness is assumed. For instance, it does causes a problem in S09, around line 750 on `lMACHTED_FNAME` (typo intended): 

```sh
    lMACHTED_FNAME=$(grep ";${lMD5_SUM_MATCHED};" "${P99_CSV_LOG}" | cut -d ';' -f1 || true)
```

`lMACHTED_FNAME` is normally a single file path, but if there are duplicates in `P99_CSV_LOG`, the command returns multiple file paths separated by new lines. `lMACHTED_FNAME` value is then put into an array and used later on as a "single file path" but this fails (namely it does not find such path with new lines in it). This makes S09 skip processing of such files while they should normally be processed. 

There is likely other places where non-uniqueness is problematic.

**To Reproduce**

Steps to reproduce the behavior:
1. EMBA installation: default mode
2. Use the firmware available here (link valid until March 14): https://1drv.ms/u/c/6832161d9629c783/EZ37i9kSzC9Iv2DHiXgUtAcBqfzvDJH6xMliUd7_uTh9Sg?e=LeJw3s
3. Extract the archive _before_ starting emba (otherwise this cause another issue: https://github.com/e-m-b-a/emba/issues/1463)
4. Start EMBA with the following parameters: sudo ./emba -p ./scan-profiles/default-sbom.emba -l <output_folder> -f /path/to/lib64

Check `csv_logs/p99_prepare_analyzer.csv`. 
- If the race condition occurred, it will contain 3 or 4 lines. 
- Otherwise, it will contain only two lines. In that case, re-run emba until the race condition occurs (should not be more than a few times)

**Expected behavior**

MD5 sums in P99 sums should be unique to avoid causing issues in other modules. 

**Screenshots**

Example of broken `csv_logs/p99_prepare_analyzer.csv` (occurred on the first run for me):

```text
/logs/firmware/lib64-2/libjitterentropy.so.2;ELF64;2's complement, little endian;AdvancedMicroDevicesX86-64;;;ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=8d1524820e3be0bf80bc111baec4329988c19054, stripped;d5ebb335b01804237703c40093f497a6;
/logs/firmware/lib64-2/libjson-c.so.4.0.0;ELF64;2's complement, little endian;AdvancedMicroDevicesX86-64;;;ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=6c879f21c380efc32c264def15ee6b860c74b8d5, stripped;0ea304346edc653ab0a1cd3df38e8335;
/logs/firmware/lib64-2/libjitterentropy.so.2.1.2;ELF64;2's complement, little endian;AdvancedMicroDevicesX86-64;;;ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=8d1524820e3be0bf80bc111baec4329988c19054, stripped;d5ebb335b01804237703c40093f497a6;
```

**Desktop (please complete the following information):**
- OS: Ubuntu 22.04 VM
- EMBA version: master branch as of 2025-02-10, commit 15e67eb92b925a9d93e696ce4728d4c9509f342e
- Installation method: default with up to date docker image (1.5.1b)

**Priority issue**

Are you already a [Sponsor]? - No

**Additional context**

m-1-k-3, 2025-02-15:
Another interesting issue you found. Quick fix is to ensure we only use one binary in the s09 and other modules:

```
    lMACHTED_FNAME=$(grep ";${lMD5_SUM_MATCHED};" "${P99_CSV_LOG}" | cut -d ';' -f1 | head -1 || true)
```

And we should fix the typo ;)

To ensure we really have only unique entries in p99 we probably need to post-process the csv for cleaning it up. this will take time ...

gluesmith2021, 2025-02-17:
> Another interesting issue you found. Quick fix is to ensure we only use one binary in the s09 and other modules:
>
> ```
>     lMACHTED_FNAME=$(grep ";${lMD5_SUM_MATCHED};" "${P99_CSV_LOG}" | cut -d ';' [...]

Instead of post-processing, it seemed easier after all to prevent the problem from the source: PR #1471

m-1-k-3, 2025-02-20:
Fixed with https://github.com/e-m-b-a/emba/pull/1471
Thanks @gluesmith2021

closed 2025-02-20 by m-1-k-3