e-m-b-a/emba #1472: QNX Support in EMBA

ttepatti, 2025-02-18:
Hello! I wanted to raise the idea of potentially adding QNX support to EMBA, to see what the maintainers thought and if it was a path worth putting energy towards.

**Is your feature request related to a problem? Please describe.**
QNX is a commercial "UNIX-like real time operating system" developed by BlackBerry. (You can view more info about the OS on their [Wikipedia page](https://en.wikipedia.org/wiki/QNX)) It's widely used as an embedded operating system, most notably within the automotive industry.

Currently, QNX-based firmware can be scanned using EMBA, but there aren't many QNX-specific detections or extractions within EMBA.

**Describe the solution you'd like**
- Add QNX OS detection to S06 - Linux distribution detection
- Add QNX IFS filesystem extraction module to "Pre-Modules"

And, as a stretch goal:
- Add QNX-specific bootloader and system startup file detection to S07
- Add QNX-specific microkernel detections to S24/S25, if possible
- Add QNX-specific CVEs/vulnerability reporting if an old/vulnerable version of QNX is detected

**Describe alternatives you've considered**
Currently, much of this QNX analysis is possible manually. If you extract QNX filesystems outside of EMBA before scanning them, EMBA still does a great job scanning through all of their binaries - because of that, these requests are mainly just 'quality of life' improvements.

**Priority issue**
Are you already a Sponsor? - No

**Additional context**
- One thing that could make IFS support particularly tricky is QNX's commercial/proprietary state. That being said, QNX has been made open source in the past and there are open source extractors for the IFS format, but I am definitely not an expert on software licensing.
- Additional details on QNX's "Image File System" (IFS) format are available on the QNX developer docs website: https://www.qnx.com/developers/docs/7.0.0/#com.qnx.doc.neutrino.building/topic/intro/intro_ifs.html
- Finding a piece of QNX firmware that is freely available and could be used for unit testing/feature testing could be difficult. Automotive companies in particular are usually somewhat restrictive with their firmware distribution.

Anyhow, thank you for reading! I'd be more than happy to help put these features together, if you think it would be a useful addition to EMBA. I'm a big fan of EMBA and poke at a lot of QNX stuff in my free time, so I'd love to see more QNX-related features added to the project.

Thanks!

m-1-k-3, 2025-02-18:
Good idea @ttepatti 

We can try to address two areas:
* Extraction: As far as I understand there are special filesystems or packing used for this kind of firmware. This should be addressed in our main extraction frameworks [unblob ](https://github.com/onekey-sec/unblob) and [binwalk](https://github.com/ReFirmLabs/binwalk). Please open dedicated issues in both projects. If they integrate it, it will automatically land in EMBA.
* Analysis: We can start straight ahead with some extracted filesystem and your knowledge where to find the juicy areas and what we should look for.

For both steps testing firmware is needed. Is there something out there in the Internet?

m-1-k-3, 2025-03-08:
Any updates on this?

ttepatti, 2025-03-13:
> Any updates on this?

Hey there! Apologies for taking a bit to reply. So, overall I really like your ideas - I'm currently collecting information on the QNX "qnx6" file system structure and notes on how binwalk extracts files from it so I can prepare an issue to file with the main binwalk project. Hopefully I'll have that done soon!

The notes I'm collecting are:
- What kind of QNX file systems are there
- Are there any open source implementations that give examples of how to mount and/or read them?
- Are there any specific file signatures that could be used for identification of these filesystem types?
- How does binwalk currently deal with these filesystems, and how could its analysis or extraction be improved?

Part of that process was building the latest binwalkv3 to ensure I wasn't using anything outdated, but unfortunately that's led me to the hardest issue to solve: Finding public binaries!

This ties into your final comment, about how for testing and development, firmware is needed. I've been working to track down some publicly-released QNX firmware, but haven't found any examples quite yet. Much of the content I've worked with is automotive and has been dumped from the flash of specific parts, so it's likely not able to be easily accessed or redistributed.

It seems like [QNX have recently loosened up their license requirements](https://www.qnx.com/products/everywhere/) to encourage open source development and hobbyist use, which includes releasing a Raspberry Pi board support package. This could be a great starting place depending on the licensing and any asterisks it has, so I'm going to dig into it a bit further. I'll be sure to update again when I have more leads!

m-1-k-3, 2025-03-13:
Good to hear. Keep us in the loop ...

ttepatti, 2025-03-17:
So, as a potential solution to getting firmware for analysis, I've done a bit more digging into QNX's new non-commercial QNX SDP 8.0 license:

https://blackberry.qnx.com/en/products/qnx-everywhere/licensing

While it _seems_ like this could be a good solution to getting a stock QNX firmware image, I'll be 100% honest and say I'm not a licensing expert. It seems like working on open source tool development is in line with the non-commercial license's bullet point about open source projects:

> A Non-Commercial license may be right for you if you want to:
> - Develop open-source software (OSS) that is interoperable with QNX software, provided you make the resulting OSS publicly available at no charge.

Buut, I honestly can't say for sure.

As for poking around with the new SDP and testing things out:

There's a great guide on getting up-and-running with the new SDP 8.0 by creating a headless Pi, available on the QNX developer blog: https://devblog.qnx.com/set-up-a-headless-qnx-raspberry-pi/

And they reference these more technical QNX GitLab instructions as well: https://gitlab.com/qnx/quick-start-images/raspberry-pi-qnx-8.0-quick-start-image/-/wikis/Getting-started?ref=devblog.qnx.com

But from what I can tell, the biggest restriction seems to be around redistribution. From reading the non-commercial license, it doesn't seem like you would be able to re-distribute compiled binaries, at least from what I can tell?

So if you wanted to work with the compiled QNX Raspberry Pi image to develop EMBA functionality, it seems like each developer would have to:
- request a non-commercial SDP license
- download the QNX software center
- download the example Raspberry Pi image to their machine

And finally, they could perform analysis/development using it 😓 Which is definitely non-optimal.

There's a new QNX spokesperson named John that seems to be their public point of contact for their new open source license details - I may try reaching out to him to get some clearer wording regarding the licensing and how it may be compatible with open source tool development.

That's all I have now for an update - slow progress, but in the right direction! 🙂 I'll let you know if I hear anything back.

Thanks!

m-1-k-3, 2025-03-18:
Thank you for your update. The process of getting a QNX up and running does not look that complicated. Nevertheless, currently I have no resources to do it by myself. :( You work is highly appreciate

ArkaprabhaChakraborty, 2026-01-08:
Hey! I would like to pick this up!