e-m-b-a/emba #1476: Limit emulation output size (on top of existing time limit)

gluesmith2021, 2025-02-20:
**Is your feature request related to a problem? Please describe.**

Emulation output files can be hugely oversized. For example, I noted this down a few months ago, from analyzing a Linux-based firmware:

Largest files under `logs/s115_usermode_emulator`:

```text
-rw-r--r--  1 user user 6701138328 Nov 12 15:16 qemu_tmp_yes.coreutils.txt
-rw-r--r--  1 user user 1017832515 Nov 12 15:15 stracer_yes.coreutils.txt
-rw-r--r--  1 user user  446668800 Nov 12 13:55 qemu_initx_yes.coreutils.txt
-rw-r--r--  1 user user  111947562 Nov 12 13:55 qemu_init_yes.coreutils.txt
```

Largest files under `logs/html-report/s115_usermode_emulator`:

```text
-rw-r--r--  1 user user 21993364120 Nov 12 18:27 qemu_tmp_yes.coreutils.html
-rw-r--r--  1 user user  6594096960 Nov 12 18:28 stracer_yes.coreutils.html
-rw-r--r--  1 user user   725250981 Nov 12 18:28 qemu_init_yes.coreutils.html
```

That is a whopping **37.5 GB** solely for emulating the common and very small `yes` utility (here named `yes.coreutils`).

Other executables such as `sulogin` and `nettle-lfib-stream` (a random number generator) generated about 1 GB each. 

In all those files, there only seem to be a few kB worth of data (about 2 to 4 kb in `.txt` files), and everything only records a flooded stdout (with a repeated strings or random data as with `nettle-lfib-stream`). This wastes disk space and makes reporting modules longer to run. 

**Describe the solution you'd like**

**On top** of the running time limit (20 seconds by default), I'd welcome an output size limit on every `qemu` run. That is, not a limit on `.txt` file size, but rather on the output of individual instances of `qemu`.

It seems that a few kilobytes should be enough in most cases, but this could be 64 or 1024 kB to be sure not to miss anything. Indeed, in some instances, such as `env` or `printenv` (that outputs running emba code....) 1 MB would do fine.

**Describe alternatives you've considered**

- Reducing the emulation time limit (e.g. 2 seconds instead of 20 seconds)
  - **Workaround**: Seems it can't be overridden from a scan profile, must modify `helpers_emba_defaults.sh`
  - **Bad**: effectively reduces size for the largest output files, but they are still way too large 
  - **Bad**: as storage becomes faster this problem will come back 
  - **Potentially bad**: May prevent programs from being properly emulated if they really need a few seconds for instance
- Patch S115 to limit output size
  - **Bad**: piping into `head -c 64k` would limit size in some instances but I think the broken pipe (once size limit is reached) is what totally messed up most outputs (typically no ouptut at all, even sometimes garbage). This is even after disabling buffering on stdout using `stdbuf`, otherwise there is no output at all.

**Priority issue**

Are you already a [Sponsor]? No

**Additional context**

m-1-k-3, 2025-02-20:
In s115 you will find a lot of dirty corners of EMBA. ;)

* The size limit is an issue we should address. I thought we have this addressed already [here](https://github.com/e-m-b-a/emba/blob/732574ad583a2e6b052ceb0b9ffcdbe3102d036f/modules/S115_usermode_emulator.sh#L720). Looks as this is not working as expected.
* The runtime is on 20 secs as we had some stuff that took a while to run until it prints the interesting data. In combination with threading and slow machines the 20secs were a quite fine limit. As it is in the defaults helper you could try to set it up via a scan-profile.

gluesmith2021, 2025-02-20:
Indeed, changing run time is not a solution. 

I didn't know there was already something to limit file sizes, so it is probably broken. 

Here's what I found:

Binary names are extracted with

```sh
mapfile -t lCRITICAL_FILES_ARR < <(find "${LOG_PATH_MODULE}"/ -xdev -type f -size +"${KILL_SIZE}" -print0 2>/dev/null |xargs -r -0 -P 16 -I % sh -c 'basename % 2>/dev/null| cut -d\. -f1 | cut -d_ -f2' || true)
```  

Then corresponding processed are found with `pgrep -f "${lEMULATOR}.*${lKILLER}"`  where `${lKILLER}` is meant to be the binary name

If we run the first command on above filenames:
- `qemu_tmp_yes.coreutils.txt`
- `stracer_yes.coreutils.txt`
- `qemu_initx_yes.coreutils.txt`
- `qemu_init_yes.coreutils.txt`

It returns respectively:
- `tmp`
- `yes`
- `initx`
- `init`

So:
- In all cases but `stracer`, it returns the wrong binary name (tmp, initx, init)
  - Process is not killed, as tmp, initx and init do not appear in process command line
- `emulate_strace_run()` does not call check_disk_space_emu()
  - stracer is not killed either
  - Can't tell though if this would make sense to call check_disk_space_emu() from emulate_strace_run() as well
  - If the initial `find | ...` is fixed for the three other cases, it will break for stracer. Might require to call the file "stracer_tmp_..." or something similar.

m-1-k-3, 2025-02-21:
Please check #1477. I addressed the following issues:
* fix the stracer_, qemu_tmp and qemu_initx_ naming issue in process name detection
* it should be possible to set the size limit in the defaults and in the scanning profile via QEMU_KILL_SIZE
* fix the naming issue in cases like 'asdf.qwertz"
* Default file size is now limited to 10M (probably we can also limit this to 1 or 5M)

closed 2025-02-21 by m-1-k-3

gluesmith2021, 2025-02-21:
Wonderful! and that was quick! 

I'll give it a go. Thank you!

m-1-k-3, 2025-02-21:
As I have not found a testing firmware for this case I hope it is fixed now. If not just ping me again

PR #1477 (2025-02-21): S115 #1476
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

Logs are growing on differents user-mode emulation tasks - see #1476

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

process detection fixed
kill size limit setup via defaults and via scan-profile possible
kill size limit to 20M
fix process name detection in cases like "yes.coreutils"
max-depth in find command for further search limitations
closes #1476

* **Does t [... cut]