e-m-b-a/emba #1497: Failed to configure transport while installing

aL-tst, 2025-03-07:
Hi,
I see this error while installing, do you have any clue to it? Appreciate it!

**To Reproduce**
Use the firmware available here: https://github.com/e-m-b-a/emba/releases/tag/v1.5.1-rise-from-the-dead

- OS: Ubuntu 24.04.1 (VMware 6.11.0)
- EMBA version: v1.5.1
- Docker version 28.0.1, build 068a01e

[image]

BenediktMKuehne, 2025-03-07:
Can you download the docker image manually?
Try:
`sudo docker compose pull`

Might be a network issue

m-1-k-3, 2025-03-08:
looks for me like a network issue. Please reopen if not

closed 2025-03-08 by m-1-k-3

aL-tst, 2025-03-10:
Thanks.
It appeared to be a network issue, the installation finally succeeded after several attemps.