e-m-b-a/emba #1508: Supported ISAs?

billy-1-brown, 2025-03-12:
Which ISAs (ARM, x86, etc.) are currently supported by Emba?

m-1-k-3, 2025-03-12:
Hi @billy-1-brown 

 EMBA only partly depends on dedicated architectures. So, for the most modules the target architecture does not matter. On other modules we support only some architectures. For example the module S13 needs a supported architecture and checks for it here https://github.com/e-m-b-a/emba/blob/fbe1811232deef637fa1de90dea10d65f9b93410/modules/S13_weak_func_check.sh#L63

Nevertheless, as this module depends on objdump and readelf it supports every architecture (with some little tweaks) which is supported by objdump. If you are missing your architecture it should be quite easy to implement it.

Some further examples:
Module S15 uses r2 and supports every architecture that r2 supports. 
Module s16 is the same just with Ghidra
Module s115 needs a static compiled qemu and supports currently some architectures but if needed it can be tweaked easily.
The L modules are depending on a pre-compiled kernel. I think we have support for some MIPS, ARM and x86 architectures. You can check the wiki for further details.

Hope this helps

m-1-k-3, 2025-03-19:
Please reopen if needed

closed 2025-03-19 by m-1-k-3