e-m-b-a/emba #1519: kali-linux-2025-1-release

m-1-k-3, 2025-03-20:
New Kali Linux is out: https://www.kali.org/blog/kali-linux-2025-1-release/

We need to test EMBA on it

**Testcases:**

Testfirmware: [DLink DIR300](https://ftp.dlink.de/dir/dir-300/archive/driver_software/DIR-300_fw_revb_214b01_ALL_de_20130206.zip)

- [ ] Default/docker installation working with current docker image
  - [ ] `./installer.sh -d` finished without errors
  - [ ] dependency check (`./emba -d 1`)
  - [ ] EMBA run with profile quick-scan
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)
- [ ] Docker base image build
  - [ ] `sudo docker-compose build --no-cache --pull` finished without errors
  - [ ] dependency check (`./emba -d 2`)
  - [ ] EMBA run with profile quick-scan
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)
- [ ] Full installation working on Kali Linux
  - [ ] `./installer.sh -F` finished without errors
  - [ ] dependency check (`./emba -d 1`)
  - [ ] EMBA run in dev mode with options -s -z -S -D -E -t -W -Q
  - [ ] EMBA run in dev mode with options -s -z -S -D, -c, -E, -t, -W -Q

**Priority issue**
YES

torabi12, 2025-03-20:
Install was fine:

[image]

Dependency chack is also error free:

[image]

These scans were also OK:

`$ sudo ./emba -l ~/DIR300_quick -f ~/DIR300B5_FW214WWB01.bin -p ./scan-profiles/quick-scan.emba -X 2.14 -X DLink -Z DIR300 -N DIR300B5_FW214WWB01.bin`

`$ sudo ./emba -l ~/DIR300_default -f ~/DIR300B5_FW214WWB01.bin -S -p ./scan-profiles/default-scan.emba -X 2.14 -X DLink -Z DIR300 -N DIR300B5_FW214WWB01.bin`

torabi12, 2025-03-25:
The 5th test was done today:

[image]

torabi12, 2025-03-26:
and the last result:

[image]

m-1-k-3, 2025-03-28:
The updated docker base image `embeddedanalyzer/emba:1.5.2a` is up for testing. Just do a `docker pull` on it. Afterwards you need to update your image entry in your local docker-compose file and it should be ready to go.

torabi12, 2025-03-28:
Thanks, I did it.
`./emba -U` drops this for me:

[image]

but I will execute some tests soon.

torabi12, 2025-03-28:
[image]

default-scan profile for the DIR300 FW. If you help what to check inside the logs please share with me.
But the scan was OK.

m-1-k-3, 2025-03-29:
> default-scan profile for the DIR300 FW. If you help what to check inside the logs please share with me. But the scan was OK.

Looks good, The results should match the results from the same scan with the older base image and the older Kali host environment.

m-1-k-3, 2025-03-29:
Looks as they changed the entropy plot here https://github.com/ReFirmLabs/binwalk/commit/09693d38ceeb1be7fb0f0006479022594072ced9

torabi12, 2025-03-29:
Without strict mode it was OK:

```
=================================================================
The firmware bin file check module conducts an analysis of binary firmware files, extracting key details like checksums and entropy, generating visual entropy graphs and identifying specific firmware types such as UEFI/BIOS, AVM or D-Link encrypted

[*] Entropy testing with binwalk ... 

[*] Details of the firmware file:
    -rw------- 1 1001 1001 3.5M Feb  6  2013 /firmware
    /firmware: data

00000000  5e a3 a4 17 00 00 00 20  00 00 00 00 73 69 67 6e  |^...... ....sign|
00000010  61 74 75 72 65 3d 77 72  67 6e 34 39 5f 64 6c 6f  |ature=wrgn49_dlo|
00000020  62 5f 64 69 72 33 30 30  62 35 00 00 5e a3 a4 17  |b_dir300b5..^...|
00000030  00 00 00 24 00 37 a0 20  77 42 bc 45 81 16 45 f7  |...$.7. wB.E..E.|
00000040  53 7d 85 f0 b9 df 4b b3  64 65 76 3d 2f 64 65 76  |S}....K.dev=/dev|
00000050  2f 6d 74 64 62 6c 6f 63  6b 2f 32 00 74 79 70 65  |/mtdblock/2.type|
00000060  3d 66 69 72 6d 77 61 72  65 00 00 00 5d 00 00 00  |=firmware...]...|
00000070  02 0c 18 35 00 00 00 00  00 00 4e 03 fc 00 8c 66  |...5......N....f|
00000080  f8 58 2f a1 b8 d4 a5 bf  31 9b 56 ad 7e 8f ad 62  |.X/.....1.V.~..b|
00000090  75 bc 8a 42 ed 6b c3 59  69 f7 e1 36 b6 9b 2a f5  |u..B.k.Yi..6..*.|

[*] SHA512 checksum: 9d998f3a6dbcd5c025029e28f6652861c1f351c907689e7d6365a7a81fd546c27ca6244b09ce666f30db82e12703c2760ff0e90524da1c3bfb3f174ee0d6cc39

[*] Entropy of firmware file:
    7.989077 bits per byte.

[*] Visualized firmware file (first 2000 bytes):
```

I started a default-scan with -S and I have the same:

[image]

m-1-k-3, 2025-04-02:
The updated docker base image embeddedanalyzer/emba:1.5.2b is up for testing. Binwalk installation currently fixed to commit 2916ddfed802c61b84f4567c9c1734d69c2e320d according to #1530

m-1-k-3, 2025-04-05:
done

closed 2025-04-05 by m-1-k-3