e-m-b-a/emba #1521: Error detected - status code 127 EMBA 1.5.2

dennisarriola, 2025-03-21:
**Describe the bug**
Wasn't able to install EMBA 1.5.2. default docker.
Throws: 
Command:  echo -e "Command:  ${ORANGE}${BASH_COMMAND}${NC}"
Location:  ./installer/I05_emba_docker_image_dl.sh, line 57
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 57 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 372 -> main -d

**To Reproduce**
Steps to reproduce the behavior:
1. Clone repository via https://github.com/e-m-b-a/emba.git
2. Run "sudo ./installer.sh -d"

**Expected behavior**
Default Docker must be installed properly

**Desktop (please complete the following information):**
- OS:
[image]
- EMBA version: v1.5.2
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - N

**Additional context**
Please logs here: [emba_logs_error_installation.txt](https://github.com/user-attachments/files/19384838/emba_logs_error_installation.txt)

Thank you very much..

m-1-k-3, 2025-03-21:
looks as there is some issue with "docker compose". Could you please check if "docker compose" is available on your system?

m-1-k-3, 2025-03-21:
I checked your logs and no automatic docker installation was performed on your system. On a plain Kali 2025.1a this installation should be done automatically. I would recommend to do it manually now. Probably you have installed it somehow before and something is missing. Check the needed steps here: https://github.com/e-m-b-a/emba/blob/84b1246f288ac3f3c81b9aa5408afabfc5e6839c/installer.sh#L291

Afterwards a restart of the installer should work.

I added another warning message to the installer here https://github.com/e-m-b-a/emba/pull/1523

dennisarriola, 2025-03-24:
Hello @m-1-k-3 

Thank you for your quick response and apologies for late reply (no work on weekends).

I managed to install docker-compose and performed manual installation of emba following "sudo docker-compose build emba" command as described in the installation guide.
However, the result was still NG.
Please see screenshot of the error below:
[image]
Could you please help me to resolve this?

I attached the logs for your better investigation: 
[emba_manual_installation_error.txt](https://github.com/user-attachments/files/19412775/emba_manual_installation_error.txt)

Thank you again!

m-1-k-3, 2025-03-24:
Looks as your sytem is trying to use https and failes with it. Have you checked your /etc/apt/sources.list and moved to http only? Then a apt-get update and try again.

dennisarriola, 2025-03-25:
Hello @m-1-k-3 ,

Thank you for your response.

Yes, I tried updating etc/apt/sources.list and moved to http only and performed apt-get update and try again but the result is the same.
Please see:
[image]

[emba_manual_installation_error_http_only.txt](https://github.com/user-attachments/files/19439862/emba_manual_installation_error_http_only.txt)

I think the error is on the repository but I just don't have idea how to resolve this.

m-1-k-3, 2025-03-25:
From my point of view this is an issue with your system in combination with the used kali mirror. Probably just try again?

dennisarriola, 2025-03-26:
Hello @m-1-k-3,

Thank you for your response!

I was able to install the tool successfully already!
I deleted all files related to emba e.g. docker images and source code and the start fresh.
I made sure that docker-compose was installed.
I think this was the reason why on my first try, it wasn't successful due to missing docker-compose.
Kindly see the ongoing scan:
[image]

Thank you again for your support.

closed 2025-03-26 by dennisarriola

dennisarriola, 2025-03-31:
Hello @m-1-k-3 

Good day!

I was just wondering how long does it usually take for EMBA tool to finish the quick scan using default scan profile? Below is the current status on my scan:

[image]

Thank you.

m-1-k-3, 2025-03-31:
Good morning @dennisarriola 

 The runtime totally depends on your machine specs and the thing you are analyzing. There is only one last module running. You can check the running module with `helpers/running_modules.sh` and you should take a look at the docker logs. You can find the docker id in the beginning of your current log-dashboard. Just do something like `sudo docker logs <your docker container id> -f` With this you should be able to see the complete output of EMBA.

dennisarriola, 2025-04-02:
Hi @m-1-k-3,

Thank always for answering my inquiries!
I understand your explanation. It took 7 days to finish the scan. I guess I need faster PC.
There were a lot of built-in profiles unders scan-profiles folder. 
I only used the "default-scan.emba", planning to try all profiles.

Thank you again!

m-1-k-3, 2025-04-02:
It finished ... thats fine as there was no big error from EMBA :) How big is your firmware and what are your system specs? The default scan profile is using extended binary analysis and this takes time.

PR #1523 (2025-03-21): Improve installer warning on missing docker installation
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

installer is not warning on not usable docker installation

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

Warns now - closes #1521 

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

* **Other information**: