e-m-b-a/emba #1563: Emba no longer analyses directories

chconil, 2025-04-22:
EMBA used to be able to analyse directories containing multiple FW images. Current version somehow fails to explore them.

./emba -D -F -y -l /home/builder/embalogs/emba -f **/tmp/fwbin**
└─p99_prepare_analyzer.csv only contains the list of firmware files, not expanded
└─p02_firmware_bin_file_check.csv only contains a dir
Entity;data;Notes;
Firmware path;/home/builder/embalogs/emba/firmware/fwbin;NA;

./emba -D -F -y -l /home/builder/embalogs/emba -f **/tmp/fwbin/image.rootfs.ubifs**
└─p02_firmware_bin_file_check.csv contains the full file of files of the ubifs
└─p99_prepare_analyzer.csv only contains the list of all files within the ubifs

[+] EMBA version: 1.5.2
[+] Run inside the official docker container (+ git pull )

m-1-k-3, 2025-04-22:
Can you provide a testing firmware?

m-1-k-3, 2025-04-24:
btw I do not recommend using the developer mode with `-D and -F` options. This mode is deprecated and is not tested and maintained anymore.

m-1-k-3, 2025-04-24:
And probably I got your problem now. Please check https://github.com/e-m-b-a/emba/pull/1562

chconil, 2025-04-24:
Thanks, having a look right now

m-1-k-3, 2025-04-29:
Please reopen if needed

closed 2025-04-29 by m-1-k-3