e-m-b-a/emba #1577: EMBA stalls when it hits Module Progress 52

KR9SIS, 2025-05-06:
## Issue Description
When I run EMBA, then it runs until the MODULES Progress counter hits 52/71. When it hits that mark then I believe it starts hanging because said counter does not increment further, even after being allowed to run for over 6 hours.

I am currently trying to debug it myself by finding where the progress bar is updated in emba/helpers/helpers_status_bar.sh within the update_box_modules() function and am trying to work my way from there to find why it stops updating.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode) EMBA installation was default mode. See bottom of post for exact installation steps taken.

2. Use the firmware available here:[ EX820v.bin ](https://cpe-firmware.siminn.is/builds/tp-link/ex820v/EX820v_UP_2023-11-12_13.15.49.bin)

3. Start EMBA with the following parameters: `sudo ./emba -l ../logs -f ./fw/EXE820v.bin -p ./scan-profiles/default-scan.emba`

4. See error**Expected behavior**I expected EMBA to finish running and then clean up instead of stalling

**Desktop (please complete the following information):**
- OS: Kali Linux 2024.04 amd64
- EMBA version: v1.5.2 or current master branch
- Installation method: default with up to date docker image

## EMBA Setup
### Docker and Docker Compose
```bash
sudo apt update
sudo apt install -y docker.io
sudo systemctl enable docker --now
sudo usermod -aG docker $USER
reboot
# Verify correct installation
docker run hello-world
```

```bash
echo "deb [arch=amd64 signed-by=/etc/apt/keyrings/docker.gpg] \
  https://download.docker.com/linux/debian bookworm stable" | \
  sudo tee /etc/apt/sources.list.d/docker.list 

curl -fsSL https://download.docker.com/linux/debian/gpg | \
  sudo gpg --dearmor -o /etc/apt/keyrings/docker.gpg

sudo apt update
sudo apt install -y docker-ce docker-ce-cli containerd.io
```

### Install EMBA

Run install script

```bash
mkdir EMBA && cd EMBA/ # Done for organisational purposes
git clone https://github.com/e-m-b-a/emba.git
cd emba
sudo ./installer.sh
```

We got notifications when running EMBA that we needed to update a few things

- EMBA repo

```bash
cd ~/EMBA/emba
git pull
```

- NVD CVE JSON feed

```bash
cd ~/EMBA/emba/external/nvd-json-data-feeds
sudo git pull
```

- cve-bin-tool

```bash
sudo pip3 install cve-bin-tool --break-system-packages
cve-bin-tool --update now
cve-bin-tool --export ~/EMBA/emba/config/cve-database.db
```

m-1-k-3, 2025-05-06:
Please check the docker logs while EMBA is running. EMBA rpints the command in the startup phase:

[image]

KR9SIS, 2025-05-06:
Ok, so I've taken quite a few screenshots of the output.
These first ones are related to something called jtr. I don't think are related to my issue, but they might make sense to someone here and be helpful.

[image]

[image]

[image]

[image]

Now we get to the cwe_checker

[image]

[image]

[image]

[image]

[image]

The cwe_checker has now finished leaving it like this for the last 10 minutes. The cursor is moving so dots are being printed out at some interval, but I see nothing further.

[image]

KR9SIS, 2025-05-06:
The two processes it says are still running are:
- S09_firmware_base_version_check
- S118_busybox_verifier

m-1-k-3, 2025-05-06:
s118 is waiting for s09. S09 could take a bit

KR9SIS, 2025-05-06:
Is the constant printing of dots and nothing else expected behavior?

m-1-k-3, 2025-05-06:
dots are good ... this means something in EMBA is alive :-D

KR9SIS, 2025-05-06:
[image]

Is it normal that running is -1? I would assume that indicates error, but the dots are still being printed out.

It passed the 52 threshold it failed on the last couple of times, but seems to stall on 55 now. Is it normal that EMBA takes over 3 hours when run on a single 64 bit firmware image with 64GB of RAM and 12 cores?

I'm currently looking at S9 and where it's printing out the dots but as of yet I don't see any places which might create an endless loop.

m-1-k-3, 2025-05-06:
RUNNING -1 is nice ... EMBA is faster as EMBA is :-D
I need to check on that.
The dots are coming from a helper function that is constantly checking the processes. This one here https://github.com/e-m-b-a/emba/blob/676f4230ba66b38e1636c6cd4c7991145f8e5ea1/helpers/helpers_emba_helpers.sh#L39 or here https://github.com/e-m-b-a/emba/blob/676f4230ba66b38e1636c6cd4c7991145f8e5ea1/helpers/helpers_emba_helpers.sh#L61 is probably responsible

KR9SIS, 2025-05-06:
I think you're definitively correct that these are likely the functions printing out the dots.
I would assume that *wait_for_pid* is the actual perpetrator in this case, because as seen from the previous picture only S09 and S118 are currently running and you mentioned that S118 waits for S09. This means there is likely something happening in S09 which causes EMBA to take such a long time (I haven't found what yet).
I'm also assuming all other modules have finished since iirc EMBA allows for twice the number of threads as CPU cores, which in my case is 12 which means there would be room for other modules to start but they don't seem to do so.

m-1-k-3, 2025-05-06:
I am also wondering why S09 is still running on your system. On my quick test it finished up after one hour of runtime:

[image]

KR9SIS, 2025-05-06:
I didn't see it in the arguments section, but is there a way for me to run EMBA with only S09? That way I can try to debug it without having to wait for all of the other parts of the code.

It's still running btw, but I'm going to kill it now since it definitively should have finished by now.

m-1-k-3, 2025-05-06:
Try -m s09

m-1-k-3, 2025-05-13:
any progress on this issue?

KR9SIS, 2025-05-13:
Not towards a solution sadly, using -m s09 worked but I have not been able to look further into it. The only thing I know is that the stall happens after bin_string_checker is called.

[tee_out.log](https://github.com/user-attachments/files/20187683/tee_out.log)

zh5j, 2025-05-17:
I think I've reproduced this using [my test firmware package](https://zxcv.nbg1.your-objectstorage.com/apks.7z)

From the main EMBA output, my progress is also stuck at 52/71 using default scan. 

I'll let my firmware package run overnight and check the logs. BRB

zh5j, 2025-05-18:
Yeah looks like in my case it's stuck at either S09 or S118
[image]
[image]
[image]

Logs as of now: (generated using `docker logs <id> 2>&1 | ansi2txt | pigz -6`)
[logs.txt.gz](https://github.com/user-attachments/files/20269977/logs.txt.gz)

As usual, many thanks for reviewing!

m-1-k-3, 2025-05-18:
> Yeah looks like in my case it's stuck at either S09 or S118

S118 is waiting for s09. I am quite sure that the issue is in S09. Which firmware are you testing?

zh5j, 2025-05-18:
> S09

https://zxcv.nbg1.your-objectstorage.com/apks.7z 

@m-1-k-3  If you're willing to try, could you please check with this package? Or if your system runs low on resources, please select several APKs from this package and run EMBA on the dir containing them. This is the same firmware I ran tests against for #1585.

zh5j, 2025-05-18:
> Yeah looks like in my case it's stuck at either S09 or S118 [image] [image] [image]

Notes on environment: Linux KVM, 112 cores, 400G RAM, all drives NVMe based, Kubuntu 22.04

3% CPU load here corresponds to roughly 3 cores fully loaded. 

As an additional observation, S24-S26 are also finishing pretty late, @KR9SIS could yuu please confirm if this was the case in your setup?

m-1-k-3, 2025-05-18:
FYI: I found another bug during testing TL-WR840Nv6_EU_0.9.1_4.19_up_boot_241230-rel57226.zip. I will address this in PR #1586

m-1-k-3, 2025-05-18:
> * [EX820v](https://cpe-firmware.siminn.is/builds/tp-link/ex820v/EX820v_UP_2023-11-12_13.15.49.bin) Original Image used
> * [TP-Link WR840N](https://www.tp-link.com/in/support/download/tl-wr840n/#Firmware) V6.2 New reproducible image
> * [TP-Link HB820](https://www.tp-link.com/us/support/download/hb810/#Firmware) V1.6 New reproducible image

all firmware images tested with my latest draft PR #1586 and the following startup command:

```
sudo ./emba -f /home/m1k3/<Firmware> -l <log dir> -y -S -m s09
```

All firmware images were running quite fine.

zh5j, 2025-05-19:
Using this test firmware [DIR300A1_FW106WWb05.bin.gz](https://github.com/user-attachments/files/20277602/DIR300A1_FW106WWb05.bin.gz), I commented out [this section](https://github.com/e-m-b-a/emba/blob/e212b5bea8835a653dc5be6a7d3d9cfb577ab403/modules/S09_firmware_base_version_check.sh#L393-L400) and got it to run through using quick scan profile, without blacklisting anything:
`screen -S emba ./emba -l /dev/shm/DIR_logs -f /root/DIR300A1_FW106WWb05.bin -p ./scan-profiles/quick-scan.emba`

Although I'm pretty sure this makes the code miss some of the threads or rules or whatever... Without commenting out this section, then the progress of S09 seems to be stuck here at "comgt.json". 

FYI, @m-1-k-3 @KR9SIS

zh5j, 2025-05-19:
> On my quick test it finished up after one hour of runtime

Honestly, I might just have been a little bit too impatient and thought something went wrong after like 20mins... although frankly it's still weird when S09 is trying to limit its multi-threading capability when the main logger shows only ~20 processes, given that my VM has 112 cores allocated. 

Anyways, I think it might be feasible to parallelize [this for loop](https://github.com/e-m-b-a/emba/blob/e212b5bea8835a653dc5be6a7d3d9cfb577ab403/modules/S09_firmware_base_version_check.sh#L174) by making everything inside run serially, then use xargs to parallelize multiple types of version checks (there are ~500 JSON files being processed in this loop). Just a crude idea :)

Anyway thanks for reviewing!

m-1-k-3, 2025-05-19:
Good to hear that it is working now. Within the mentioned loop we parallelize the further binary checking over [here](https://github.com/e-m-b-a/emba/blob/e212b5bea8835a653dc5be6a7d3d9cfb577ab403/modules/S09_firmware_base_version_check.sh#L384). Further improvements will be possible but I think we need to rework more areas in the module. If you like you can give it a try ...

KR9SIS, 2025-05-19:
> As an additional observation, S24-S26 are also finishing pretty late, [@KR9SIS](https://github.com/KR9SIS) could yuu please confirm if this was the case in your setup?

Thank you for your patience
S24 was 7 minutes and 5 seconds
S25 was 6 minutes and 59 seconds
S26 was only took 34 seconds

KR9SIS, 2025-05-19:
Commenting out the relevant section also fixed it at my end, I'll leave the issue open in case you want to discuss it further. Thank you very much for all of your help.

m-1-k-3, 2025-05-19:
> Commenting out the relevant section also fixed it at my end, I'll leave the issue open in case you want to discuss it further. Thank you very much for all of your help.

I will check the code ...

m-1-k-3, 2025-05-20:
The pid protection mechanism was the very old version and we missed to migrate it. I have updated it now in PR #1586 
Please check if the current functionality is working. The threading by itself will be improved soon.

zh5j, 2025-05-21:
Using [this firmware image](https://github.com/user-attachments/files/20277602/DIR300A1_FW106WWb05.bin.gz) I confirm S09 issue has been fixed:

Commit e212b5b
`screen -S emba ./emba -l /dev/shm/DIR_logs_e212b5b -f /root/DIR300A1_FW106WWb05.bin -S -m s09`
note: test didn't finish even after running overnight, so I hit Ctrl-C in screen

Logs (emba.log)
```
[!] Tue May 20 21:43:26 HKT 2025 - Pre-checking phase started on Tue May 20 21:43:26 HKT 2025
    Firmware binary path: /firmware
[*] Tue May 20 21:43:26 HKT 2025 - P02_firmware_bin_file_check starting
[*] Tue May 20 21:43:27 HKT 2025 - Quest container finished
[*] Tue May 20 21:43:28 HKT 2025 - P02_firmware_bin_file_check finished
[*] Tue May 20 21:43:29 HKT 2025 - P50_binwalk_extractor starting
[*] Tue May 20 21:43:36 HKT 2025 - P50_binwalk_extractor finished
[*] Tue May 20 21:43:36 HKT 2025 - P55_unblob_extractor starting
[*] Tue May 20 21:43:36 HKT 2025 - P55_unblob_extractor finished
[*] Tue May 20 21:43:36 HKT 2025 - P60_deep_extractor starting
[*] Tue May 20 21:43:36 HKT 2025 - P60_deep_extractor finished
[*] Tue May 20 21:43:37 HKT 2025 - P65_package_extractor starting
[*] Tue May 20 21:43:38 HKT 2025 - P65_package_extractor finished
[*] Tue May 20 21:43:38 HKT 2025 - P99_prepare_analyzer starting
[*] Tue May 20 21:43:39 HKT 2025 - P99_prepare_analyzer finished
[!] Tue May 20 21:43:40 HKT 2025 - Pre-checking phase ended on Tue May 20 21:43:40 HKT 2025 and took about 0 days and 00:00:17 
[!] Tue May 20 21:43:40 HKT 2025 - Testing phase started on Tue May 20 21:43:40 HKT 2025
    Firmware path: /logs/firmware
[*] Tue May 20 21:43:40 HKT 2025 - S09_firmware_base_version_check starting
[-] Wed May 21 09:03:39 HKT 2025 - Test ended on Wed May 21 09:03:39 HKT 2025 and EMBA failed in docker mode!
```

Commit 5daab8e
`screen -S emba ./emba -l /dev/shm/DIR_logs_5daab8e -f /root/DIR300A1_FW106WWb05.bin -S -m s09`

Logs (emba.log)
```
[!] Wed May 21 09:05:06 HKT 2025 - Pre-checking phase started on Wed May 21 09:05:06 HKT 2025
    Firmware binary path: /firmware
[*] Wed May 21 09:05:06 HKT 2025 - P02_firmware_bin_file_check starting
[*] Wed May 21 09:05:06 HKT 2025 - Quest container finished
[*] Wed May 21 09:05:08 HKT 2025 - P02_firmware_bin_file_check finished
[*] Wed May 21 09:05:08 HKT 2025 - P50_binwalk_extractor starting
[*] Wed May 21 09:05:15 HKT 2025 - P50_binwalk_extractor finished
[*] Wed May 21 09:05:15 HKT 2025 - P55_unblob_extractor starting
[*] Wed May 21 09:05:15 HKT 2025 - P55_unblob_extractor finished
[*] Wed May 21 09:05:15 HKT 2025 - P60_deep_extractor starting
[*] Wed May 21 09:05:15 HKT 2025 - P60_deep_extractor finished
[*] Wed May 21 09:05:15 HKT 2025 - P65_package_extractor starting
[*] Wed May 21 09:05:17 HKT 2025 - P65_package_extractor finished
[*] Wed May 21 09:05:17 HKT 2025 - P99_prepare_analyzer starting
[*] Wed May 21 09:05:18 HKT 2025 - P99_prepare_analyzer finished
[!] Wed May 21 09:05:18 HKT 2025 - Pre-checking phase ended on Wed May 21 09:05:18 HKT 2025 and took about 0 days and 00:00:15 
[!] Wed May 21 09:05:19 HKT 2025 - Testing phase started on Wed May 21 09:05:19 HKT 2025
    Firmware path: /logs/firmware
[*] Wed May 21 09:05:19 HKT 2025 - S09_firmware_base_version_check starting
[*] Wed May 21 09:06:35 HKT 2025 - S09_firmware_base_version_check finished
[!] Wed May 21 09:06:35 HKT 2025 - Testing phase ended on Wed May 21 09:06:35 HKT 2025 and took about 0 days and 00:01:32 
[!] Wed May 21 09:06:35 HKT 2025 - Reporting phase started on Wed May 21 09:06:35 HKT 2025
[*] Wed May 21 09:06:35 HKT 2025 - F02_toolchain starting
[*] Wed May 21 09:06:36 HKT 2025 - F02_toolchain finished
[*] Wed May 21 09:06:36 HKT 2025 - F05_qs_resolver starting
[*] Wed May 21 09:06:37 HKT 2025 - F05_qs_resolver finished
[*] Wed May 21 09:06:37 HKT 2025 - F10_license_summary starting
[*] Wed May 21 09:06:39 HKT 2025 - F10_license_summary finished
[*] Wed May 21 09:06:39 HKT 2025 - F15_cyclonedx_sbom starting
[*] Wed May 21 09:06:50 HKT 2025 - F15_cyclonedx_sbom finished
[*] Wed May 21 09:06:50 HKT 2025 - F17_cve_bin_tool starting
[*] Wed May 21 09:10:09 HKT 2025 - F17_cve_bin_tool finished
[*] Wed May 21 09:10:09 HKT 2025 - F50_base_aggregator starting
[*] Wed May 21 09:10:12 HKT 2025 - F50_base_aggregator finished
[!] Wed May 21 09:10:13 HKT 2025 - Test ended on Wed May 21 09:10:13 HKT 2025 and took about 0 days and 00:05:10 
[!] Wed May 21 09:10:14 HKT 2025 - Test ended on Wed May 21 09:10:14 HKT 2025 and took about 0 days and 00:05:20 
```

Thanks, appreciate your amazing productivity @m-1-k-3 !

zh5j, 2025-05-21:
Commit de9df7b
`screen -S emba ./emba -l /dev/shm/DIR_logs_de9df7b -f /root/DIR300A1_FW106WWb05.bin -S -m s09`

emba.log
```
[!] Wed May 21 09:23:45 HKT 2025 - Pre-checking phase started on Wed May 21 09:23:45 HKT 2025
    Firmware binary path: /firmware
[*] Wed May 21 09:23:45 HKT 2025 - P02_firmware_bin_file_check starting
[*] Wed May 21 09:23:46 HKT 2025 - Quest container finished
[*] Wed May 21 09:23:47 HKT 2025 - P02_firmware_bin_file_check finished
[*] Wed May 21 09:23:48 HKT 2025 - P50_binwalk_extractor starting
[*] Wed May 21 09:23:55 HKT 2025 - P50_binwalk_extractor finished
[*] Wed May 21 09:23:56 HKT 2025 - P55_unblob_extractor starting
[*] Wed May 21 09:23:56 HKT 2025 - P55_unblob_extractor finished
[*] Wed May 21 09:23:56 HKT 2025 - P60_deep_extractor starting
[*] Wed May 21 09:23:56 HKT 2025 - P60_deep_extractor finished
[*] Wed May 21 09:23:57 HKT 2025 - P65_package_extractor starting
[*] Wed May 21 09:23:58 HKT 2025 - P65_package_extractor finished
[*] Wed May 21 09:23:58 HKT 2025 - P99_prepare_analyzer starting
[*] Wed May 21 09:24:00 HKT 2025 - P99_prepare_analyzer finished
[!] Wed May 21 09:24:00 HKT 2025 - Pre-checking phase ended on Wed May 21 09:24:00 HKT 2025 and took about 0 days and 00:00:18 
[!] Wed May 21 09:24:00 HKT 2025 - Testing phase started on Wed May 21 09:24:00 HKT 2025
    Firmware path: /logs/firmware
[*] Wed May 21 09:24:00 HKT 2025 - S09_firmware_base_version_check starting
[*] Wed May 21 09:24:08 HKT 2025 - S09_firmware_base_version_check finished
[!] Wed May 21 09:24:09 HKT 2025 - Testing phase ended on Wed May 21 09:24:09 HKT 2025 and took about 0 days and 00:00:27 
[!] Wed May 21 09:24:09 HKT 2025 - Reporting phase started on Wed May 21 09:24:09 HKT 2025
[*] Wed May 21 09:24:09 HKT 2025 - F02_toolchain starting
[*] Wed May 21 09:24:10 HKT 2025 - F02_toolchain finished
[*] Wed May 21 09:24:10 HKT 2025 - F05_qs_resolver starting
[*] Wed May 21 09:24:11 HKT 2025 - F05_qs_resolver finished
[*] Wed May 21 09:24:11 HKT 2025 - F10_license_summary starting
[*] Wed May 21 09:24:14 HKT 2025 - F10_license_summary finished
[*] Wed May 21 09:24:14 HKT 2025 - F15_cyclonedx_sbom starting
[*] Wed May 21 09:24:26 HKT 2025 - F15_cyclonedx_sbom finished
[*] Wed May 21 09:24:26 HKT 2025 - F17_cve_bin_tool starting
[*] Wed May 21 09:27:56 HKT 2025 - F17_cve_bin_tool finished
[*] Wed May 21 09:27:56 HKT 2025 - F50_base_aggregator starting
[*] Wed May 21 09:28:00 HKT 2025 - F50_base_aggregator finished
[!] Wed May 21 09:28:00 HKT 2025 - Test ended on Wed May 21 09:28:00 HKT 2025 and took about 0 days and 00:04:18 
[!] Wed May 21 09:28:01 HKT 2025 - Test ended on Wed May 21 09:28:01 HKT 2025 and took about 0 days and 00:04:26 
```

m-1-k-3, 2025-05-21:
> Thanks, appreciate your amazing productivity [@m-1-k-3](https://github.com/m-1-k-3) !

Thank you for using EMBA and for pushing this topic. The S09 performance is quite amazing now :)

closed 2025-05-21 by m-1-k-3