e-m-b-a/emba #1585: How to parallelize 'Binary firmware deep extractor' ?

wuzhongjie1997, 2025-05-14:
First of all, huge thanks for open-sourcing this awesome project, it aids a lot in our firmware security research! 

Issue: 
emba running for 7 days in the 'extraction' phase

Firmware description: 
~200 APK files (no rootfs, just APKs) totalling ~5GiB, zipped

Hardware description:
Linux KVM, with 128 cores, 256G ram, 400G disk

OS version: 
Ubuntu Jammy 22.04

Observations:
'deep extract' module seems to be runing serially, analyzing files one by one. This results in extremely low CPU load (1-2%). 
From 'top -cd 1' and sorting by PID it seems a lot of time is consumed by 'sleep' processes and 'du -sh /log'.

Question:
Is it possible to parallelize the extraction, such that my CPU cores get more efficiently utilized?

Any help would be much appreciated!

m-1-k-3, 2025-05-14:
Hi @wuzhongjie1997 ,

 good to have you as EMBA user ... quite powerful environment you have :)

The deep-extractor is not fully threaded. The extraction process is always threaded (see [here](https://github.com/e-m-b-a/emba/blob/b1a36d4bcdf3c565ac3b697abd756a8d2c62a580/modules/P60_deep_extractor.sh#L228) or [here](https://github.com/e-m-b-a/emba/blob/b1a36d4bcdf3c565ac3b697abd756a8d2c62a580/modules/P60_deep_extractor.sh#L363)). As the main bottleneck is the bash looping we probably can improve the performance if we are going to move the complete calling part into a threaded function. I will give it a try. Hope you can test it and can give me then feedback if this improved your situation.

wuzhongjie1997, 2025-05-14:
attached logs for reference: (file names and paths are censored)
[logs.txt.gz](https://github.com/user-attachments/files/20208211/logs.txt.gz)

m-1-k-3, 2025-05-14:
@wuzhongjie1997 can you provide some testing firmware?

m-1-k-3, 2025-05-15:
Please check the updates with your firmware and give me an overview of the old and new performance

closed 2025-05-15 by m-1-k-3

wuzhongjie1997, 2025-05-15:
Many thanks for your effort, unfortunately the company server is busy running the mentioned scan, i'll try my best to check it out within a week. 

The firmware package i've been scanning is from a customer project and has to be kept confidential. I'll try extracting APKs from an old phone and benchmark on that.

wuzhongjie1997, 2025-05-15:
One more question, does EMBA monitor (and better, self-regulate) ram usage in VM? asking because my server is currently running low on physical ram, from what i've seen, reserving 2GiB per VCPU seems to be an absolute necessity as my current test VM's ram usage peaked at 93%

m-1-k-3, 2025-05-16:
> As I don't quite understand the internals of this project, when trying to benchmark and differentiate between two commits, do I need to download the docker container again, or is it sufficient to just modify the scripts? ( git checkout )

This depends on the latest docker-compose configuration. You can alway check the available EMBA docker images with `sudo docker images`. Then check the docker-compose configuration for the image entry. The versions need a match

[image]

m-1-k-3, 2025-05-16:
> One more question, does EMBA monitor (and better, self-regulate) ram usage in VM? asking because my server is currently running low on physical ram, from what i've seen, reserving 2GiB per VCPU seems to be an absolute necessity as my current test VM's ram usage peaked at 93%

The RAM usage heavily depends on the firmware you are testing and the modules you are running. For example if you are having big binaries and EMBA is analyzing them via cwe-checker, Ghidra, R2 it needs a lot of RAM. Please also check the Prerequisites in our Wiki: https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites

zh5j, 2025-05-17:
Test firmware:

https://zxcv.nbg1.your-objectstorage.com/apks.7z
~900MiB, 235 APK files extracted from a Sony XZ Premium (only system apps included)
Extracted package to directory and ran EMBA on directory, to avoid binwalk runs on large file

Environment:

Linux KVM, 112 cores, 400G RAM, 160G disk
Kubuntu 22.04
EMBA commit e212b5bea8835a653dc5be6a7d3d9cfb577ab403 installed according to README.md

Observations:

From what I see now, using the latest master, the extracting process is indeed very fast, finishing in 10 minutes. 
I'd say the original issue has been properly resolved. 

Logs:

[logs.txt.gz](https://github.com/user-attachments/files/20263714/logs.txt.gz)
[logerr.txt.gz](https://github.com/user-attachments/files/20263713/logerr.txt.gz)

Remarks:

Great thanks for your time @m-1-k-3 !

m-1-k-3, 2025-05-17:
Thanks for reporting this issue and verifying the fix.

zh5j, 2025-05-17:
[image]

[image]

Further observations:

Looks like in my test case with bunch of APKs (and even more small files nested within, 87807 in total in my test firmware), [this loop](https://github.com/e-m-b-a/emba/blob/96d8f82b1eafdc2a188221ebec1d58f965b009f3/modules/P60_deep_extractor.sh#L77-L82) is using lots of CPU in a single thread. I've verified this by adding debug identifiers before & after this loop. (see first screenshot of `docker exec -it <id> top -cd 1`, note the PID 1 takes ~10min CPU time). When adding a print to the loop, I see that it's processing about 130 files per second, taking just over 10 minutes to finish the loop. 

Question:
From my crude understanding, this is where we're `Populating backend data`, so i'm wondering if this would be easily parallelized?  

Thanks in advance for reviewing!

PR #1587 (2025-05-15): Deeper extractor update
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Feature

* **What is the current behavior?** (You can also link to an open issue here)

Deep extractor is quite slow - see #1585 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

Should improve/close #1585

* **Does this PR introduce a breaking change?** (What changes might users need to make in their application due to this PR?)

It is now possible to switch between binwalk and unblob for the deep extractor.
Note: The EMBA default extract [... cut]