e-m-b-a/emba #1589: Missing SQLite CVE database updates

busby666, 2025-05-20:
**Describe the bug**
New installation/Update in the Elementary check after the Network Connection check. I consistently get the following message:
[*] Elementary:
    user permission - ok
    Docker compose version - docker compose ok
    Docker-compose EMBA image version - ok
    Configuration directory - ok
    Python virtual environment - ok
    NVD CVE database in JSON format - ok
    **SQLite CVE database update in config directory - not ok
    Missing SQLite CVE database updates - Check update instructions**

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. After the install
3. ./emba -U
4. Error Messages
  5. SQLite CVE database update in config directory - not ok
      Missing SQLite CVE database updates - Check update instructions

**Expected behavior**
Would not expect this error on a new installation.

**Platform**
Kali Linux Version kali-linux-2024.4-installer-amd64.iso
Added the NEW/Updated KEY and had updated the image before this test.  if I can provide anything please let me know.

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2024.03
- EMBA version: v1.5.0 or current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2025-05-20:
Please check https://github.com/e-m-b-a/emba/wiki/EMBA-update

busby666, 2025-05-20:
Brand new install of Kali and same thing.
Then did a reboot.
After reboot "./emba -U" gives the same issue.

Do not see how the steps in the URL above would address this?

m-1-k-3, 2025-05-20:
The database is included in the EMBA docker base image. Nevertheless, this database is outdated and so you can update it by yourself on your machine

[image]

busby666, 2025-05-20:
sudo -s
pip3 install cve-bin-tool --break-system-packages
ve-bin-tool --update now

ERROR    CVEDB - Unable to fetch EPSS, skipping EPSS.  

then a lot of errors; how should I address?

m-1-k-3, 2025-05-20:
As there is no critical need for the cve-bin-tool update mechanism, we can wait for an official fix

m-1-k-3, 2025-05-25:
Tested it and it looks as it is working as expected. Probably there was some hickup somewhere ...

[image]