e-m-b-a/emba #1614: Feature Request: Reuse Existing SBOM for CVE Rechecking Without Full Rescan

Jeff-Rowell, 2025-06-17:
**Is your feature request related to a problem? Please describe.**

Yes. EMBA generates a comprehensive SBOM and SBOM+VEX JSON in the `/SBOM` directory within the log directory specified with the `-l` argument as part of its full firmware scans, which is great. However, the scans (especially using the `default-scan.emba` or `default-scan-long.emba` profiles) can take 1–2 hours or sometimes more to complete depending on the size of the firmware file and number of kernels and components found. When new CVEs are published or existing ones are updated, we currently have to rerun the entire deep scan just to check if the firmware is affected.

This is inefficient when the underlying firmware components haven’t changed, especially when scanning hundreds of firmware images with EMBA. In many workflows, we only want to re-evaluate known components against the latest CVE data, not repeat the full static/dynamic analysis.

**Describe the solution you'd like**

I’d like EMBA to support a feature where it can take a previously EMBA-generated SBOM (e.g., JSON or JSON+VEX format from a prior scan) and recheck the components listed in it against up-to-date CVE databases.

Ideally, this would be implemented via a new command-line option such as:

```
./emba.sh --reuse-sbom SBOM/EMBA_cyclonedx_sbom.spdx
```

Or passing in the VEX SBOM to get an updated VEX report:

```
./emba.sh --reuse-sbom SBOM/EMBA_cyclonedx_vex_sbom.json
```

This mode would:
- Parse the existing SBOM for all identified software components and versions.
- Fetch the latest CVE data from the usual feeds (e.g., NVD, EPSS, GAD, etc.).
- Match SBOM entries against new or changed CVEs.
- Output a new report including newly discovered vulnerabilities.

This could also be implemented as a new module that EMBA users can optionally configure in a custom scan profile (i.e., use `default-scan-long.emba` the first scan then use `default-rescan.emba` for future/continuous re-scans). The module would implicitly skip/blacklist any modules it doesn't need to use to regenerate an updated SBOM.

**Describe alternatives you've considered**

The only current workaround is to rerun the full EMBA scan every time we want updated CVE coverage. This is time and resource-intensive and doesn’t scale well when scanning hundreds of firmware images.

We've tried to feed the EMBA produced SBOMs into tools like `grype` and it did not return nearly as many CVEs as originally found in the SBOM+VEX JSON. Additionally we've tried to just run `cve-bin-tool` to try to get an updated vulnerability report using the SBOM, but encounter several errors as shown below:

[image]

... several more invalid license errors, then this:

[image]

Another idea was to extract the SBOM and write an external tool to recheck CVEs manually, but this seems redundant when EMBA already has the SBOM structure and CVE-matching capabilities.

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

N

**Additional context**

This feature request aligns with industry trends toward SBOM reuse and incremental CVE scanning.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2025-06-19:
I will give some further notes on your initial request ...
> Yes. EMBA generates a comprehensive SBOM and SBOM+VEX JSON in the `/SBOM` directory within the log directory specified with the `-l` argument as part of its full firmware scans, which is great. [...]

Are you also aware of the SBOM scanning profiles. These should speed up the SBOM generation massively.

> When new CVEs are published or existing ones are updated, we currently have to rerun the entire deep scan just to check if the firmware is affected.

Or just the SBOM scan which should speed up things.

> 
> This is inefficient when the underlying firmware components haven’t changed, especially when scanning hundreds of firmware images with EMBA. 

Absolutely. I am impressed that you are scanning so many firmware images with EMBA. Probably you will find a lot of issues with cornercases that we have never seen before. If you are interested please run EMBA with '-S' in strict mode to get more errors. Also if something does not work as expected on some firmware ... report it :)

> In many workflows, we only want to re-evaluate known components against the latest CVE data, not repeat the full static/dynamic analysis.

Absolutely.

> 
> **Describe the solution you'd like**
> 

See your PR :) I will take a review later on.

> 
> This mode would:
> 
> * Parse the existing SBOM for all identified software components and versions.
> * Fetch the latest CVE data from the usual feeds (e.g., NVD, EPSS, GAD, etc.).

Probably we need to split it up. On the on hand we have the default CVE data in the docker base image. Updating this would be more complicated. Nevertheless, you can export a database and this will be used automatically in your next test:

```
cve-bin-tool --export ./config/cve-database.db
```
You  could to this every night automatically via cron and EMBA will then use it.

> **Describe alternatives you've considered**
>
> The only current workaround is to rerun the full EMBA scan every time we want updated CVE coverage. This is time and resource-intensive and doesn’t scale [...]

Grype is optimized for syft generated SBOMs. So, I think we need to adjust the SBOM before getting the best Grype results.

> Additionally we've tried to just run `cve-bin-tool` to try to get an updated vulnerability report using the SBOM, but encounter several errors as shown below:

Probalby the same issue as with grype. We need to adjust the json to improve results or check the errors in detail.

> 
> Another idea was to extract the SBOM and write an external tool to recheck CVEs manually, but this seems redundant when EMBA already has the SBOM structure and CVE-matching capabilities.

Absolutely

m-1-k-3, 2025-06-20:
> > Are you also aware of the SBOM scanning profiles. These should speed up the SBOM generation massively.
> > >
> 
> I have played around with the `quick-sbom.emba` and `default-sbom-vex.emba` profiles. I tested those out on a few firmwares and they were faster, but missed some results in some cases. Sometimes the firmwares I test have multiple linux kernels so I really need to do the deep extraction and then rescan the SBOM with a fresh CVE export from cve-bin-tool.

I think this is a bug. The default SBOM profile should also handle multiple kernels. Do you have a testcase for me?

closed 2025-06-28 by m-1-k-3