e-m-b-a/emba #1694: Missing lua detection case

gluesmith2021, 2025-08-27:
**Describe the bug**
Some instances of `lua` are not detected

**To Reproduce**
Steps to reproduce the behavior: default mode EMBA, but can't share firwmares nor files that are statically linked with lua library (validation: binaries export all `lua*_*` functions and import none). So see below instead.

**Expected behavior**
`lua` is detected

**Screenshots**
Binaries contain the following string (copied from extracted strings under `s09_firmware_base_version_check/strings_bins`)

`$LuaVersion: Lua 5.2.2  Copyright (C) 1994-2013 Lua.org, PUC-Rio $$LuaAuthors: R. Ierusalimschy, L. H. de Figueiredo, W. Celes $`

**Desktop (please complete the following information):**
- OS: Ubuntu
- EMBA version: 1.5.2 release
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor][1]? No

m-1-k-3, 2025-08-27:
I think we can fix this via adding another detection rule after the following in here https://github.com/e-m-b-a/emba/blob/c87c26e20bbbe6e2a14b3c338c13185f2c8be9ca/config/bin_version_identifiers/lua.json#L10

should look like this:

```
"LuaVersion:\\ Lua\\ [0-9](\\.[0-9]+)+?\\ \\ Copyright\\ .*"
```

Please check if this is working as expected and initiate a PR afterwards.

Thank you

gluesmith2021, 2025-08-28:
While this will likely fix my case, could it break others? I mean, is there always `Lua:` or `LuaVersion:` before the rest?

m-1-k-3, 2025-08-28:
Please do not replace the current rule. We should add this additional rule to ensure nothing breaks.

gluesmith2021, 2025-09-03:
Sorry for the delay: yes, it works fine.  And thank you for the fix.